gyroflow/gyroflow #849: Fix remaining Sony issues

AdrianEddy, 2024-06-26:
- [ ] Wrong stabilization in some files with IBIS, possibly related to low light (C0763, C1327)
- [ ] Dynamic EIS doesn't work properly (C1784, C1398)
- [ ] Warping in some cases, possibly related to IBIS/OIS/EIS but it could also be the lens profile mismatch with EIS mesh (C1877, C1363)
- [ ] Frame readout time and FOV smoothness adjustments, when shooting in APS-C mode (C0161, C2302)

rgaufman, 2024-07-11:
Sometimes the output is very good and near perfect, but sometimes you get this:

Original: https://drive.google.com/file/d/1piTvnoM8rIRYrAF_RnLUsRGnOvn2Os4Y/view?usp=sharing
Stabilised: https://drive.google.com/file/d/1pi6mRdhUOCdxFE-Jc39WkN4S6b-Yu6OZ/view?usp=sharing

Notice the weird stepping, not sure what is causing it.

Screenshot of settings:
[image]

AdrianEddy, 2024-07-11:
hmm, that file stabilizes perfectly for me and I don't get these steppings. Can you reset all gyroflow settings to defaults and try again? (in Advanced on bottom right)

rgaufman, 2024-07-11:
Strange, must be my settings. After resetting it's perfect, thank you!

AdrianEddy, 2024-07-29:
Fixed `Wrong stabilization in some files with IBIS, possibly related to low light (C0763, C1327, C0454)` in ba109a500c4b180a67f394065595258588fec994, turned out `first_frame_timestamp` can be different per frame in the metadata

AdrianEddy, 2024-07-30:
Added support for `FocalPlaneDistortion` data in a07f45226784b141534705a793ad41f518f910cd

nonuser666, 2024-10-18:
Hi!
I found a bug when I use Gyroflow plugin with Davinci Resolve.

When video recorded in vertical and changed the output size to 16:9 in the plugin, and ticked the Use plugin RoD in the settings panel.

Its gives me a distorted output.

Everything up to date:
OS: win 10
DR: 19.2
GF: latest
Gpu: 3090 drive latest

Camera: sony fx30
Video mode 4k highest format and Bitrate 23.9 fps

If it helps i can link a video from the bug by tomorrow. 

In horizontal its works great! Thank you for the great work!

AdrianEddy, 2024-10-18:
Did you set Input rotation in the plugin?

nonuser666, 2024-10-18:
Yes i tried.

AdrianEddy, 2024-10-18:
@nonuser666 I tried it and in this case, you have to go to Clip Attributes and set `Image Orientation` to 0, and then in plugin (in Fusion) set the Output size to your timeline size (eg. 1920x1080) and `Video rotation: 90` and `Input rotation: 0`. 
It can't be done automatically because Resolve doesn't report the value in "Image orientation" to the plugin, so the user must have to take care of it

nonuser666, 2024-10-18:
Okay.
Tomorrow i will give it a try and let you know.
Thank you!

MoonStruckHorrors, 2024-10-19:
Hello, facing a weird issue in multiple files where in stabilised files, the frame sways left and right noticeably even though the original video is near static (Apart from the vibrations ofcourse).

Not sure if the gyro data produced by the camera is at fault or the stabilisation algorithm.

Input: https://drive.google.com/file/d/1KCkcndWWWT9axmHy0GqACqS4ZAP7SXpJ/view?usp=drive_link 
Output: https://drive.google.com/file/d/1gJIJt-jpdY4Xm9161TxKtObM4_QVOqb8/view?usp=drive_link

Tried with default settings as well.

Settings:

[image]

nonuser666, 2024-10-19:
> @nonuser666 I tried it and in this case, you have to go to Clip Attributes and set `Image Orientation` to 0, and then in plugin (in Fusion) set the Output size to your timeline size (eg. 1920x1080) [...]

Worked! thank you! 
I made a short tutorial if someone has the same problem.
https://youtu.be/ZXtmdWn4y_w

AdrianEddy, 2024-10-19:
@MoonStruckHorrors Change `Integration method` (in Motion data section) to "Complementary". The VQF algorithm is more stable with more data, but with short clips it can "overreact" to the first few seconds until it settles. "Complementary" doesn't have this issue

AdrianEddy, 2024-10-19:
@nonuser666 I noticed in your video you're using the official version. The new Sony support is available in the dev build of the app and plugin from here https://gyroflow.xyz/devbuild/

nonuser666, 2024-10-19:
I will test it. Thank you for the feedback!

MoonStruckHorrors, 2024-10-20:
> @MoonStruckHorrors Change `Integration method` (in Motion data section) to "Complementary". The VQF algorithm is more stable with more data, but with short clips it can "overreact" to the first few seconds until it settles. "Complementary" doesn't have this issue

Thanks @AdrianEddy 

Works great with `Complementary`.

AdrianEddy, 2024-11-11:
You have to set your timeline to "Scale entire image to fit" 
https://docs.gyroflow.xyz/app/video-editor-plugins/davinci-resolve-openfx#vertical-video-or-different-aspect-ratio

recfilm, 2024-12-22:
Hi, I have tried different ways and still got bad results.
I have a file that was accidentally recorded vertically. Video information: 270 degree rotation.
In davinci resolve > clip attributes I set it to 0 degrees and it works fine, but I can't enter good project settings with gyroflow to Davinci.
Every time the image is either incorrectly rotated, stretched or narrowed.. Project settings in davinci set to horizontal with a resolution of 1920x1080.
[image]

AdrianEddy, 2024-12-22:
did you set `Input rotation` and `Video rotation` to 0 in the plugin?

recfilm, 2024-12-22:
Yes, I tried. I finally did it in inspector.
The problem is that Fusion still reads the source file as 1080x1920, and after stabilization the timeline only shows part of the image.
I tried to get around this with compound clip or new fusion composition, but gyroflow stabilization does not work in this configuration.

AdrianEddy, 2024-12-22:
> Fusion still reads the source file as 1080x1920

It shouldn't do that if you set "Orientation" to 0 in `Clip attributes`, make sure you did that for the source file. In any case, that's not related to gyroflow

You should be able to use exiftool to reset the rotation flag in mp4: `exiftool -rotation=0 video.mp4`. For more info see [here](https://superuser.com/questions/1307158/change-rotation-flag-in-mp4-video-without-losing-metadata)

keithkml, 2025-02-22:
I posted this on Discord, but posting here in case it's useful to keep issues and example videos in one place.

I use Sony a6700 with Sony 200-600mm f/5.6-6.3 lens. I always use both body and lens stabilization systems, as I do a mix of photography and videography. In some (most?) videos I am seeing odd distortion that looks almost like extreme, coarse-grained heat distortion. I've [uploaded the videos (before/after)](https://drive.google.com/drive/folders/1Q4EVl3b_aGox6McQ0PIpJUDq4T_Qx7Cs) in case it's helpful in diagnosing. 

I also built Gyroflow from source with the `opencl_undistort.cl` patch that @AdrianEddy [uploaded to Discord](https://discord.com/channels/797044698682228736/797045396425408522/1325156079515992175). That patch attempts to change the "order of operations of ibis/ois/eis/lens correction." I don't think it helped, at least not much - that video is in the Google Drive folder above, called `C0137_stabilized_with_patch.mp4`.

Let me know if there's any other information you need to diagnose this issue! It's basically preventing me from using Gyroflow, though I do keep trying :)

nonuser666, 2025-04-02:
> Yes, I tried. I finally did it in inspector. The problem is that Fusion still reads the source file as 1080x1920, and after stabilization the timeline only shows part of the image. I tried to get [...]

Could you solve the problem?

I am facing the same issue. 
I made the YouTube tutorial. 

[image]

The output Image is cropped in for some reason. 
I did everything like before.

malykhcom, 2025-09-01:
It's very strange. Gyroflow adds shaking to this video. After experimenting with different settings, I found that the shaking is related to the "rolling shutter correction." Disabling this setting almost eliminates the shaking in some scenes, such as at 1:15, but introduces it in others, such as at 0:03. Other videos recorded with this camera (Sony a7c) and this lens (Carl Zeiss Jena 80/2.8) are stabilized by Gyroflow without this issue.

Video and lens profile: [https://drive.google.com/drive/u/0/folders/1vNM-jAcKti9_pii4laz4R1RhceWiwBaB](https://drive.google.com/drive/u/0/folders/1vNM-jAcKti9_pii4laz4R1RhceWiwBaB)

closed 2026-09-04 by AdrianEddy