gyroflow/gyroflow #850: Import Lens properties from PTGUI to Gyroflow. It is a much better solution for creating lens templates for a 200/250degree fisheye entaniya lens for RED/Arri  or super wide angle lens like gopro MAX2 mode

AndriiShramko, 2024-06-26:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Import Lens properties from PTGUI to Gyroflow. It is a much better solution for creating lens templates for a 200/250degree fisheye [entaniya  lens](https://products.entaniya.co.jp/en/list/hal-220-series/ ) for RED/Arri  or super wide angle lens like gopro MAX2 mode than to use a gyroflow grid.

For 10 years, I have professionally created lens defish templates for stitching 360 videos from fisheye lenses using a PTGUI software.  Who can create a script for importing lens settings from a PTGUI project or template file into gyroflow?  For now it is notpossible stabilize a fisheye video in gyroflow without correct lens template, but it is not possible create it with gyroflow techniques. 

Here is a sample PTGUI files with one fisheye defished lens. Also a gopro12MAX2lens fisheye video for stabilization test in gyroflow.  
https://drive.google.com/drive/folders/1qipgi3MXqq3VmOnhcKzxYZjhOy5xu5Zv?usp=sharing 

I will share with you a professional PTGUI license to do it. Who can try to do it? What ill be the money cost? Thanks.
Best regards Andrii Shramko

Lets be friends on Linkedin https://www.linkedin.com/in/andrii-shramko/ 

PS First of all I would like to thank Adrian for solving the problem with multi in out points when using Davinci plugin. Now it works perfectly! Here is an example of stabilized through gyroflow 16k 360 60fps video https://www.youtube.com/watch?v=GNoMRknTIxA here is 12k file for Apple Vision Pro https://drive.google.com/drive/folders/11g1i0FG9mOhl61SQ1DvGT-tJZ7XfkHgP?usp=sharing

AdrianEddy, 2024-06-26:
Does PTGui support exporting this model to anything else? maybe STMap?

AndriiShramko, 2024-06-26:
here is a DB file from PTGUI where PTGUI can store lens profiles https://drive.google.com/file/d/1NaxRxfQuxxIKTMm2nWAiFCSwOpi7fbqG/view?usp=sharing

AndriiShramko, 2024-06-27:
[image]
Here are lens data from gyroflow and PTGUI  / Can you tell me what is what and I will try moving a settings from ptgui by copy/past

AdrianEddy, 2024-06-27:
It won't work that way, it has to be implemented, but the good news is that it's not too difficult
I don't know any ETA though, I'm focused on getting 1.6.0 release done right now, it will take some time

AndrewHazelden, 2024-07-15:
> It won't work that way, it has to be implemented, but the good news is that it's not too difficult

Hi @AdrianEddy and @AndriiShramko.  

I just came across this Gyroflow issues post today. I have the exact same workflow needs as Andrii does. 

I wanted to make a short note for other people who are also interested in this PTGui <> Gyroflow lens distortion model data interchange topic:

PTGui uses the same base lens distortion model as the open-source panoramic stitching tools Hugin/Panotools do. The letters  "PT" in the PTGui product name come from its shared panoramic R&D history that was built around the original Panotools "PTSticher" syntax standard.

If I'm not mistaken Gyroflow has a set of "[ptlens](https://github.com/gyroflow/gyroflow/blob/6ba6c322fc1426c47f8429d760a303b6914625a9/src/core/stabilization/distortion_models/ptlens.rs#L5)" distortion model resources that implement the panotools centric A/B/C lens distortion parameters based upon the[ lensfun implementation](https://github.com/lensfun/lensfun/blob/e78e7be448c81256cce36a5a37ddc229616c0db7/libs/lensfun/mod-coord.cpp#L5).

[image]

[image]

---

I will connect with @AndriiShramko, to follow up on his requests in this issue post. I am going to see if I can help create some scripted tools to carry out the PTGui to Gyroflow data translation workflow he is interested. At the same time, it looks like I will need to start learning Rust programming skills this month! :)

AdrianEddy, 2024-07-17:
The missing piece in gyroflow is implemeting the "fisheyefactor" parameter, which is described here https://hugin.sourceforge.io/docs/manual/Fisheye_Projection.html and probably these functions there https://github.com/lensfun/lensfun/blob/db4db53f4e89b46845ecfdc3d61a150018035ab4/include/lensfun/lensfun.h.in#L2634-L2654

AndrewHazelden, 2024-07-17:
Ah. Good to know. 👍