gyroflow/gyroflow #852: Gyroflow terribly slow on iMac 5K Retina with AMD Radeon Pro

Giammi, 2024-06-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4

### What operating system are you using?

macOS Ventura 13.6.7

### What GPU are you using?

Radeon Pro 580 8Gb

### What happened?

I tried several export settings, but there’s no way I can go faster than 0.2-0.3 fps. Exporting a couple of minutes clip results in hours of rendering.

Specifically I’m trying to export a GoPro 11 footage in H265, 5312x4648. Using CPU gives slightly better results (0.5 fps). GPU is absolutely unusable. It works, but as written above is too slow.

[image]
[image]
[image]
[image]

AdrianEddy, 2024-06-29:
did you try the dev build? https://gyroflow.xyz/devbuild/

Giammi, 2024-06-29:
> did you try the dev build? https://gyroflow.xyz/devbuild/

Tested it right now and no improvements. Same slowness.

AdrianEddy, 2024-06-29:
try different "Device for video processing"
also, send gyroflow.log file, it's in "Show package contents" -> Contents -> MacOS

Giammi, 2024-06-29:
> try different "Device for video processing" also, send gyroflow.log file, it's in "Show package contents" -> Contents -> MacOS

Ok, so: originally I was using "OpenCL Apple AMD Radeon Pro". Very slow (0.2 fps), but works. I tried "WGPU AMD Radeon Pro Metal" and the whole iMac got stuck. After a while (I'd say a good 3 minutes), without clicking anything, the rendering started but with 0.0 fps. I hence aborted the process.

Going back to OpenCL, works again, slow as usual.

Oh, one thing that maybe is important to mention. When I select GPU, I get prompted that the Pixel format is not supported by GPU.

[image]

If I select YUV420P (that's the original one), I get this error:

[image]

With NV12 or P010LE it works, but slow as described above.

Nevertheless, here's the log:
[gyroflow.log](https://github.com/user-attachments/files/16042185/gyroflow.log)

AdrianEddy, 2024-06-29:
That GPU doesn't support 10-bit encoding, so I'm afraid your only option is encoding on CPU (or recording 8-bit in gopro)
There's nothing I can do about it

Giammi, 2024-06-29:
> try different "Device for video processing"
> 
> also, send gyroflow.log file, it's in "Show package contents" -> Contents -> MacOS

Oh, I see. At least I know what is the reason. Thanks a lot for your time!

closed 2024-06-29 by AdrianEddy