gyroflow/gyroflow #853: Side-by-side viewer

adamzimnyy, 2024-06-30:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

I would love to have a side-by-side viewer of unstabilised/stabilised footage 

[image]

AdrianEddy, 2024-07-06:
Done in 1479a19. Available in latest dev build https://gyroflow.xyz/devbuild/

`Ctrl+click` on the Stabilization overview or use the `v` keyboard shortcut. Split view is possible only when stabilization overview is enabled. The stabilized video view is lower quality, because it's a cutout from the texture with stabilization overview to make this work efficiently and to make the implementation easier. Rendering both windows separately (for full quality) is too much work to be worth it

closed 2024-07-06 by AdrianEddy

latenitefilms, 2024-07-06:
Might be a good addition to the plugins too?

AdrianEddy, 2024-07-06:
Very hard to do in plugins, would require a very significant rewrite of the rendering pipeline which is not something I want to touch