gyroflow/gyroflow #854: Davinci's gyroflow plugin preview is slow when using AMD graphics card

thisisjin, 2024-07-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

DaVinci Resolve/OpenFX v1.3.0 for Windows

### What operating system are you using?

win10 enterprise 22H2 19045.4529

### What GPU are you using?

5700XT

### What happened?

For the same video clip, using the Gyroflow plug-in in davinci, INTEL 12400F the Nvidia 3060TI graphics card can quickly complete the playback preview ，There won't be any lag, but when theAMD R5 5600X ,5700XT clicks on the playback preview, the playback speed is only 0-1 frames. I consulted the instruction document and tried to find the answer through Google, but found no solution.
[image]

AdrianEddy, 2024-07-02:
Did you try the dev build of the plugin https://gyroflow.xyz/devbuild/?

thisisjin, 2024-07-02:
> Did you try the dev build of the plugin https://gyroflow.xyz/devbuild/?

thanks，Wait, let me try now.

thisisjin, 2024-07-02:
> Did you try the dev build of the plugin https://gyroflow.xyz/devbuild/?

Thank you! The problem is solved
[image]

closed 2024-07-02 by AdrianEddy