gyroflow/gyroflow #855: Clipped blacks when rendering GoPro 12 Log footage on Windows.

ronhp, 2024-07-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

v1.5.4

### What operating system are you using?

Windows 10 Pro 22H2

### What GPU are you using?

Intel Core i9-9900K

### What happened?

Rendering GoPro 12 Log footage results in the final clip showing clipped/darkened blacks. This only happens on my windows PC, but not on the friends Mac.

I have tried all different encoders and export settings in Gyroflow, but the result is the same.

I have attached a photo of how it looks inside Gyroflow BEFORE rendering (looks how it should).
Then one AFTER rendering with Gyroflow. You can see how cooked the blacks are.

[image]
[image]

### Relevant log output

```shell
https://www.dropbox.com/scl/fo/64s3ywi1xy7jgygnpmign/AKv7wVidbbsIfZYltAjmhpg?rlkey=azg0veyy23kp9865yuubpjoo8&dl=0
```

AdrianEddy, 2024-07-03:
this is a playback issue, not rendering issue. What player are you using?
In MPC-HC, it's under `Renderer Settings -> Output range` .
In DaVinci Resolve, it's in `Clip Attributes -> Data levels`

ronhp, 2024-07-03:
> this is a playback issue, not rendering issue. What player are you using? In MPC-HC, it's under `Renderer Settings -> Output range` . In DaVinci Resolve, it's in `Clip Attributes -> Data levels`

I see... im playing back in Premiere Pro. How would I find this in Premiere?

AdrianEddy, 2024-07-03:
hmm, I'm not sure in Premiere, can you send me the original gopro file? and which codec you rendered to

ronhp, 2024-07-03:
Here is the download to the original file: https://www.dropbox.com/s/garlppo1lnmfda9/GX011685.MP4?dl=0

It's quite a long video sorry as it was a FPV shot from a friend.

The desired output codec is ProRes HQ or 4444.

I just took a look at Resolve now, and can confirm selecting Full Range does work. I'd love to know how to do this in Premiere though.

ronhp, 2024-07-03:
Also, I still don't understand why this only happens with shots that are rendered through windows.

All the clips rendered via Mac don't have this playback issue, no matter what OS plays it back.

AdrianEddy, 2024-07-03:
This is a very complex issue and a huge mess across operating systems, players, encoders, decoders and codecs. There's no simple answer or solution. See #75 for more discussion about this issue
I'll check your file soon

ronhp, 2024-07-03:
Jesus, what can of worms have I just opened haha.

Well, for now I'm just going to have to grade it in Resolve and use the 'Full Range' fix.

I personally think it's a Windows issue purely because I have GF renders of the same footage that were done on a Mac and they play back perfectly fine - no difference in colour.

ronhp, 2024-07-05:
> This is a very complex issue and a huge mess across operating systems, players, encoders, decoders and codecs. There's no simple answer or solution. See #75 for more discussion about this issue I'll check your file soon

Did you manage to check the file?

I ran the clip through ReelSteady. The issue does not happen at all.
I also tried with GF version 1.5.3, still no luck.
Tried with other peoples Hero 12 footage, no luck.
Tried on a friends windows computer, no luck.

How is this not more talked about? Is this not happening to a huge number of people using GoPro 12 and gyroflow on Windows?

ronhp, 2024-07-05:
Holy fuckin shit I figured it out. Kinda....

So my workaround was to simply use the OFX version of Gyroflow in Resolve, then render my clips out from resolve using DNxHR.
I was just maxing out the settings to begin with, rendering in DNxHR 444 12-bit. Then I thought - ok, I don't need 444 12-bit if I'm only using 10bit GoPro footage.

So then I went to render in regular DNxHQ... and I get the cooked blacks/incorrect range reading again.

From here I just went back to Gyroflow standalone app and rendered in DNxHR 444 and what do you know it works. Initially I only tried all other codecs in their HQ equivalents. With DNxHR, I didn't actually chose the highest one because I thought why would that even make a difference?

It's purely just a matter of using at least DNxHR 444. I still can't explain why HQ doesn't work though.

AdrianEddy, 2024-07-05:
hmm, no matter what settings I use, all the clips look exactly the same to me in Premiere.
They look differently in MPC-HC player, but both Premiere and Resolve show them all correctly (on Windows)

Did you try on the dev build? https://gyroflow.xyz/devbuild/ if not, try it and let me know your results.

Also, when using OFX plugin, the color management is entirely on the Resolve side. Gyroflow doesn't touch color at all there, and can't alter it in any way

AdrianEddy, 2024-07-05:
ah right, it already works fine in latest build, but not in official 1.5.4
Nothing to do here then, it's already fixed (probably in ffmpeg)

closed 2024-07-05 by AdrianEddy