gyroflow/gyroflow #857: Unable to use "Load for current file" "Browse" via the DaVinci Resolve/OpenFX v1.3.0 on DaVinci Resolve Studio 19 beta 4

kylemagne, 2024-07-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.54

### What operating system are you using?

MacOS

### What GPU are you using?

M1 Pro

### What happened?

I can't use this plugin normally as on DaVinci Resolve Studio 18.6.6.
It just doesn't stabilize anything.

AdrianEddy, 2024-07-05:
need more details

AdrianEddy, 2024-07-07:
I just tested 19 beta 4 and everything works as expected. I tested both Mac and Windows.
Make sure you're using the latest dev build of the plugin  https://gyroflow.xyz/devbuild/
You also probably need to change the scripting to "Local" in Resolve preferences, as per instructions

closed 2024-07-07 by AdrianEddy