gyroflow/gyroflow #858: different output of quat  between "println" and "write_out"

laocan, 2024-07-05:
https://github.com/gyroflow/gyroflow/blob/428f051905d2e61202dbc14a8c5c51d8223d3028/src/core/stabilization/frame_transform.rs#L230

I try to print the value of quat, 
`println!("y:{}, quat (final per line): {}",  y, quat);`
the output shows below:
`y:1045, quat (final per line): UnitQuaternion angle: 0.039434608038194645 − axis: (-0.5281908109625076, 0.6450554290858964, 0.5521937709007395),`

when I try to write_out with the following code:
`let tp = (y, "q_final:",  quat, "r=",r);`  
`let tp_str = format!("{:?}", tp);`
`let mut file = File::create("output.txt").unwrap(); `
`file.write_all(tp_str.as_bytes());`

the output of "write_out" below is different from the "println" output:
`y:1045, "q_final:", [-0.010413824001994781, 0.012717929904518116, 0.010887066995124817, 0.9998056202675577],`

BTW, the output of the matrix "r" is the same.

laocan, 2024-07-05:
One more question...
when convert quat to rot_matrix, 2 values mismatch:
[image]
when  convert rot_matrix to quat, the real part( q_w )  turn out to be opposite.
Is it because the real part of a quaternion does not affect the rotation?

should I add missing code as below?
`r[(2, 1)] *= -1.0; r[(1, 2)] *= -1.0;`

https://github.com/gyroflow/gyroflow/blob/428f051905d2e61202dbc14a8c5c51d8223d3028/src/core/stabilization/frame_transform.rs#L241

AdrianEddy, 2024-07-05:
In the first line you're printing axis and angle which is not the same as raw representation of the quaternion (.as_vector()) 
https://docs.rs/nalgebra/latest/nalgebra/geometry/type.UnitQuaternion.html#method.axis_angle
https://docs.rs/nalgebra/latest/nalgebra/geometry/struct.Quaternion.html#method.as_vector

> should I add missing code as below?

Yes

closed 2024-07-05 by AdrianEddy