gyroflow/gyroflow #859: Keyframes not loaded from .gyroflow file

czukowski, 2024-07-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

e23fb5f

### What operating system are you using?

Windows 11

### What happened?

Loading a .gyroflow file (saved by a release version) does not show keyframes in UI. The keyframed field (roll angle correction, in my case) has a constant value set, equal to the last keyframe from the file.

May be broken by 3d306320, though by looking at the diff I can't see the apparent reason. It works with dfcea83d (one commit before that).

If I make some keyframes in 3d306320 and save a .gyroflow file, it is then loaded correctly.

AdrianEddy, 2024-07-05:
good catch, it's because of the added `id` field
[image]

I'll fix it soon

AdrianEddy, 2024-07-05:
Should be fixed by c1acf6b5526e90911c19ff8ef9fd5c7df2fda7eb

closed 2024-07-05 by AdrianEddy

czukowski, 2024-07-08:
@AdrianEddy thanks for looking into it, I'll try to confirm soon.

By the way, if I use Gyroflow OFX plugin, do I need to take special care to use one based on the same dev version of Gyroflow, or is there no risk of something stabilizing differently due to version differences?

AdrianEddy, 2024-07-08:
You should use latest dev build of the plugin too, it's also available on https://gyroflow.xyz/devbuild/
The stabilization algorithm itself was changed since the official version

czukowski, 2024-07-08:
That's what I've been doing until now that I was testing different dev builds. I'll take note of the versions I'm using.