gyroflow/gyroflow #861: Gyroflow crashes during exporting

BlueGene00, 2024-07-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

v1.5.4 (gh2012

### What operating system are you using?

Win10

### What GPU are you using?

RTX4070 Super

### What happened?

During export of footage of DJI Avata 2, Gyroflow closes without any error message and export is aborted.

I have reinstalled the latest GPU driver and also tried the latest build of Gyroflow.

AdrianEddy, 2024-07-10:
Please include gyroflow.log file after it crashes

AdrianEddy, 2024-07-11:
Looks like you have GPU encoding disabled and rendering on CPU, is that what you want? It seems to crash in ffmpeg, so maybe just enable GPU encoding? And make sure you have latest drivers from nvidia.com

BlueGene00, 2024-07-11:
Yes, in the past I have used only CPU encoding since there was somewhere mentioned, that this will result in better quality. Since I am doing a lot of post-processing and shoot in D-Log-M, I want just perform the stabilization and not change the quality of the footage.

Latest Nvidia drivers are installed.

AdrianEddy, 2024-07-11:
the quality difference is very small, alternatively double the bitrate and render on GPU
but the best option is to use the OpenFX plugin in Resolve directly

BlueGene00, 2024-07-11:
[b6ce3d7f-53c7-40c0-bd7b-99822811295e.dmp](https://github.com/user-attachments/files/16180339/b6ce3d7f-53c7-40c0-bd7b-99822811295e.dmp)
[gyroflow.log](https://github.com/user-attachments/files/16180340/gyroflow.log)

I have tried GPU-export, it crashes again, even when I am working on the PC

AdrianEddy, 2024-07-11:
it still shows as CPU-only export in the log, maybe reset all settings to default (in Advanced on bottom right)