gyroflow/gyroflow #862: GPU accelerated encoder doesn't support this pixel format (YUV422P10LE).

rgaufman, 2024-07-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

82abf3f

### What operating system are you using?

MacOS 15.0

### What GPU are you using?

M3 Max

### What happened?

When I try to export I see this dialog but the M3 Max does support YUV422P10LE:

[image]

AdrianEddy, 2024-07-10:
The M3 chip does, but ffmpeg doesn't. I [submitted the patch](https://patchwork.ffmpeg.org/project/ffmpeg/patch/20230911160055.7538-1-adrian.eddy@gmail.com/) to ffmpeg a year ago, but it's still not merged.
There's nothing more I can do about it, it's up to ffmpeg.

closed 2024-07-11 by AdrianEddy

rgaufman, 2024-07-11:
Amazing! - it now exports at 21fps where CPU was only 1-2fps, thank you!!

AdrianEddy, 2024-07-11:
Cool
it's still not merged by ffmpeg, but I was able to ask the author of the custom ffmpeg build script to [add that patch](https://github.com/wang-bin/avbuild/commit/fed105e6130b86c9d6d71c5af77eae51301ccca5) unofficially for his builds

rgaufman, 2024-07-11:
Amazing, I appreciate it :)

rgaufman, 2024-10-09:
It seems there is a regression? - I've installed the latest nightly build and seeing this again:
[image]

AdrianEddy, 2024-10-09:
Yes I know, custom patches were applied only to ffmpeg master, not 7.1, I'm waiting for the fix
in the meantime you can use gh2133 which is the last version with ffmpeg master

rgaufman, 2024-10-09:
Ah, understood. I'm on this page but can't seem to find how to download previous builds? < https://gyroflow.xyz/devbuild - can you please point me to how I can download gh2133?

AdrianEddy, 2024-10-09:
It's on Github -> Actions -> Release #2133 -> Artifacts at the bottom 
https://github.com/gyroflow/gyroflow/actions/runs/11023995766

AdrianEddy, 2024-10-10:
ok latest dev build should be good now, please test and let me know

rgaufman, 2024-10-10:
Amazing, thank you, works a treat!