gyroflow/gyroflow #864: Incomplete Output of Video Frame Correction Information

laocan, 2024-07-16:
I tried to print and write out the correction information for each frame, but I found that: 
if the video stream is played directly, more than half of the intermediate video frames do not print or write output data; 
if played frame by frame, 10% of the frames still have no data output. 
The code I added is after this line:
https://github.com/gyroflow/gyroflow/blob/6ba6c322fc1426c47f8429d760a303b6914625a9/src/core/stabilization/frame_transform.rs#L237

laocan, 2024-07-16:
From frame 617 to frame 779, the frames with the sequence numbers shown below have no output information.
[image]

closed 2024-07-16 by AdrianEddy

laocan, 2024-07-18:
Thx!
Can you briefly describe how to render all video frames, after synchronizing and stabilizing a shaky video in the player. So that 
 I can dump all~

AdrianEddy, 2024-07-18:
hmm, not sure what you mean, just click Export?

laocan, 2024-07-18:
> This is expected, not all frames are rendered for playback because of multiple factors, eg frame processing time, audio sync and caching. If you want to dump all, you have to render the video, instead of doing it in playback

Thx!
Can you briefly describe how to render all video frames, after synchronizing and stabilizing a shaky video in the player. So that
I can dump all~

laocan, 2024-07-18:
My goal is to print out the final correction information for all frames, but currently, some frames are being skipped. Will I be able to obtain the correction information for all frames after rendering all of them?

AdrianEddy, 2024-07-18:
yes, rendering processes all frames. What do you need this for? Camera data can be exported simply using the Motion Data->Export option in the UI

laocan, 2024-07-18:
I have written my own rolling shutter correction method and want to verify its feasibility. Therefore, I need Gyroflow to obtain the correction quaternions for every line of each frame image. However, I am currently unable to get the correction quaternions for all frames because some frames are being skipped.

AdrianEddy, 2024-07-18:
ok then just render the video from gyroflow and dump then, it processes all frames. Note that it processes the 3 video planes per frame (Y, U and V) so you need to use just the first one per frame (Y channel)