gyroflow/gyroflow #865: crash when open any video files

huntkao, 2024-07-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

master code

### What operating system are you using?

windows 10

### What GPU are you using?

nvidia geforce mx550

### What happened?

Crash when open any test videos.

[gyroflow.log](https://github.com/user-attachments/files/16302540/gyroflow.log)

AdrianEddy, 2024-07-19:
Install latest drivers for your integrated Intel card, from intel.com and also make sure you have latest driver from  nvidia.com. the point is, install latest drivers for **both** GPUs

closed 2024-07-19 by AdrianEddy

huntkao, 2024-07-19:
Does CUDA version matter?

Thank you so much.

AdrianEddy, 2024-07-19:
no, but I'm pretty sure the crash is related to your integrated Intel card, not NVIDIA one

huntkao, 2024-07-19:
oh... Thanks. I will check it out.

huntkao, 2024-07-19:
Both drivers for integrated Intel card and NVIDIA card are updated. But the fresh-built gyroflow still crashs. 
The new log file u
[gyroflow.log](https://github.com/user-attachments/files/16309855/gyroflow.log)
ploaded.

AdrianEddy, 2024-07-19:
does the build from https://gyroflow.xyz/devbuild/ work?

huntkao, 2024-07-19:
yes. devbuild works.

AdrianEddy, 2024-07-19:
ok then try to rebuild everything again:
1. delete the `ext` folder
2. update rust: `rustup update`
3. install deps: `just install-deps`
4. compile and run: `just run`

huntkao, 2024-07-19:
I followed your steps. 
At the first time running, It hangs when I clicked the "dont load .gyroflow file" dialog. (no crash)
But at the later runs, it crashes immediately every time when that dialog box appears.

[gyroflow.log](https://github.com/user-attachments/files/16310799/gyroflow.log)

huntkao, 2024-07-19:
I followed your steps. 
Also I found that there was system-level Qt confliction problem.
It works awesome after I fixed all the things. 
Thanks a lot!