gyroflow/gyroflow #866: Click on a keyframe point doesn't jump to the exact frame, but with 1 frame offset,  keyframe is not selected for change.

zliy, 2024-07-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4 nightly 

### What operating system are you using?

macOS Sonoma 14.0

### What happened?

Click on a keyframe point doesn't jump to the exact frame, but with 1 frame offset,  keyframe is not selected for change.

UI scale 100%.

screen recording:

https://github.com/user-attachments/assets/63d613b0-209f-4d56-a07b-a7e33fd8c712

AdrianEddy, 2024-07-19:
I can't reproduce this, are you sure you tested on latest nightly? gh2029? Also, what's your video fps?

zliy, 2024-07-19:
[image]

AdrianEddy, 2024-07-19:
Try gh2030 once it finishes building https://github.com/gyroflow/gyroflow/actions/runs/10005347801
if that doesn't help, send me your video file and project file with these keyframes

zliy, 2024-07-19:
Still not working for me using gh2030.

Shoud able to reproduce with this video:
https://drive.google.com/file/d/1q2_LcQhikZKfauGUACjd-BnPa6A2F5hN/view?usp=drive_link

AdrianEddy, 2024-07-20:
@wang-bin Looks like mdk on macos doesn't seek exactly when using `player->seek(timestamp_ms, mdk::SeekFlag::FromStart);` It sometimes seeks to a frame before the requested timestmap, even though the requested timestamp exists exactly (but on the next frame). Any way to improve the accuracy of that seek? It doesn't seem to be a problem on other platforms, only macOS.
I checked which timestamp is returned from `player->renderVideo()` after `player->seek()` and it doesn't match. Then doing `player->seek(1, mdk::SeekFlag::FromNow | mdk::SeekFlag::Frame)` makes it seek to the next frame which has the timestamp we need

AdrianEddy, 2024-08-27:
Fixed in latest dev build https://gyroflow.xyz/devbuild/

closed 2024-08-27 by AdrianEddy