gyroflow/gyroflow #867: A few bugs prevent compiling on linux_arm64

tony-gutierrez, 2024-07-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

a3cea4eaaa8fa43fe73e27545a7c4f5cbd14b426

### What operating system are you using?

Ubuntu 22.04.4  ARM64

### What happened?

A few small bugs are preventing compiling on ARM64

Line 37 in linux.just does not account for the architecture. 
[image]

And there are a few mismatched types errors.

```
error[E0412]:** cannot find type `__va_list_tag` in crate `ffi`
   --> src/rendering/mod.rs:734:25
    |
734 | type VaList = *mut ffi::__va_list_tag;
    |                         ^^^^^^^^^^^^^ not found in `ffi`

error[E0308]:** mismatched types
   --> src/rendering/mod.rs:714:39
    |
714 |         ffi::av_log_set_callback(Some(ffmpeg_log));
    |                                  ---- ^^^^^^^^^^ expected fn pointer, found fn item
    |                                  |
    |                                  arguments to this enum variant are incorrect
    |
    = note: expected fn pointer `unsafe extern "C" fn(_, _, _, [u64; 4])`
                  found fn item `unsafe extern "C" fn(_, _, _, *mut {type error}) {ffmpeg_log}`
help: the type constructed contains `unsafe extern "C" fn(*mut c_void, i32, *const u8, *mut {type error}) {ffmpeg_log}` due to the type of the argument passed
   --> src/rendering/mod.rs:714:34
    |
714 |         ffi::av_log_set_callback(Some(ffmpeg_log));
    |                                  ^^^^^----------^
    |                                       |
    |                                       this argument influences the type of `Some`
note: tuple variant defined here
   --> /build/rustc-fgzh1o/rustc-1.75.0+dfsg0ubuntu1~bpo0/library/core/src/option.rs:577:5

error[E0308]:** mismatched types
   --> src/rendering/mod.rs:746:70
    |
746 |         let written = ffi::av_log_format_line2(avcl, level, fmt, vl, line.as_mut_ptr() as *mut i8, line.len() as i32, &mut prefix);
    |                       ------------------------                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected `*mut u8`, found `*mut i8`
    |                       |
    |                       arguments to this function are incorrect
    |
    = note: expected raw pointer `*mut u8`
               found raw pointer `*mut i8`
note: function defined here
   --> /home/thd/Downloads/gyroflow/target/release/build/ffmpeg-sys-next-ee2b32bc9e38282e/out/bindings.rs:3:212063
    |
3   | ...ze, it means\n         that the log message was truncated to fit the buffer."] pub fn av_log_format_line2 (ptr : * mut libc :: c_void , level : libc :: c_int , fmt : * const libc :: c_char , v...
    |                                                                                          ^^^^^^^^^^^^^^^^^^^

```

### Relevant log output

```shell
error[E0412]: cannot find type `__va_list_tag` in crate `ffi`
   --> src/rendering/mod.rs:734:25
    |
734 | type VaList = *mut ffi::__va_list_tag;
    |                         ^^^^^^^^^^^^^ not found in `ffi`

error[E0308]: mismatched types
   --> src/rendering/mod.rs:714:39
    |
714 |         ffi::av_log_set_callback(Some(ffmpeg_log));
    |                                  ---- ^^^^^^^^^^ expected fn pointer, found fn item
    |                                  |
    |                                  arguments to this enum variant are incorrect
    |
    = note: expected fn pointer `unsafe extern "C" fn(_, _, _, [u64; 4])`
                  found fn item `unsafe extern "C" fn(_, _, _, *mut {type error}) {ffmpeg_log}`
help: the type constructed contains `unsafe extern "C" fn(*mut c_void, i32, *const u8, *mut {type error}) {ffmpeg_log}` due to the type of the argument passed
   --> src/rendering/mod.rs:714:34
    |
714 |         ffi::av_log_set_callback(Some(ffmpeg_log));
    |                                  ^^^^^----------^
    |                                       |
    |                                       this argument influences the type of `Some`
note: tuple variant defined here
   --> /build/rustc-fgzh1o/rustc-1.75.0+dfsg0ubuntu1~bpo0/library/core/src/option.rs:577:5

error[E0308]: mismatched types
   --> src/rendering/mod.rs:746:70
    |
746 |         let written = ffi::av_log_format_line2(avcl, level, fmt, vl, line.as_mut_ptr() as *mut i8, line.len() as i32, &mut prefix);
    |                       ------------------------                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected `*mut u8`, found `*mut i8`
    |                       |
    |                       arguments to this function are incorrect
    |

                                                                            ^^^^^^^^^^^^^^^^^^^
```

AdrianEddy, 2024-07-20:
Fixed in 5abcc506f37143683d5b9a94176e83ec7db0dc0b

closed 2024-07-20 by AdrianEddy

tony-gutierrez, 2024-07-23:
Compiles and runs! 

Seems like there are a few changes needed in the **Justfile**.

Update the versions of some files for copy:
[image]

Also the deploy script seems to be trying to copy files that arent present in the deploy directory (but I do see them in the release directory):
```
cp: cannot stat '/home/thd/Downloads/gyroflow/ext/6.7.2/gcc_arm64/qml/QtQml/libqmlplugin.so': No such file or directory
cp: cannot stat '/home/thd/Downloads/gyroflow/target/deploy/libmdk-braw.so': No such file or directory
```