gyroflow/gyroflow #868: Gyroflow closes uppon loading a video file!

Yuri-jeik-Becker, 2024-07-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4

### What operating system are you using?

Windows 11 home

### What GPU are you using?

Nvidia RTX 3060 laptop

### What happened?

I have a Zephyrus G15 with both AMD and Nvidia GPUs and once Gyroflow worked fine on both but now out of nothing whenever I load a video file the program just freezes and closes, the first print is before loading a file and the second is just before it closes.
1-
[image]
2-
[image]

### Relevant log output

```shell
PS C:\Users\Yuri\Downloads\Gyroflow-windows64> .\Gyroflow.exe
PS C:\Users\Yuri\Downloads\Gyroflow-windows64> 04:17:46 [DEBUG] (1) gyroflow: [in main] Qt version: 6.6.0
04:17:46 [DEBUG] (1) gyroflow: Gyroflow 1.5.4
04:17:46 [DEBUG] (1) gyroflow: Executable path: Some("C:\\Users\\Yuri\\Downloads\\Gyroflow-windows64\\Gyroflow.exe")
04:17:46 [DEBUG] (1) gyroflow: SDK path: "C:\\Users\\Yuri\\Downloads\\Gyroflow-windows64"
04:17:46 [WARN] qrc:/src/ui/main_window.qml:38:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
04:17:46 [DEBUG] (1) gyroflow_core::filesystem: path_to_url | path:  -> Ok("")
04:17:47 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 12.5.85") ["NVIDIA GeForce RTX 3060 Laptop GPU"]
04:17:47 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("AMD Accelerated Parallel Processing") Ok("OpenCL 2.1 AMD-APP (3444.0)") ["gfx1035"]
04:17:47 [INFO] OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_d3d10_sharing cl_khr_d3d10_sharing cl_nv_d3d11_sharing cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_win32 cl_khr_external_memory_win32" } Device: NVIDIA Corporation NVIDIA GeForce RTX 3060 Laptop GPU
04:17:47 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
04:17:47 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: HWDevice::from_type AV_HWDEVICE_TYPE_CUDA, device: None
04:17:47 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
04:17:47 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia corporation nvidia geforce rtx 3060 laptop gpu
04:17:47 [WARN] qrc:/src/ui/components/NumberField.qml:56: Error: Locale: Number.fromLocaleString(): Invalid format
04:17:47 [WARN] qrc:/src/ui/components/NumberField.qml:56: Error: Locale: Number.fromLocaleString(): Invalid format
04:17:47 [WARN] qrc:/src/ui/menu/Synchronization.qml:17:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
04:17:47 [WARN] qrc:/src/ui/components/NumberField.qml:56: Error: Locale: Number.fromLocaleString(): Invalid format
04:17:47 [WARN] qrc:/src/ui/components/NumberField.qml:56: Error: Locale: Number.fromLocaleString(): Invalid format
04:17:47 [WARN] qrc:/src/ui/menu/Stabilization.qml:25:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
04:17:47 [WARN] qrc:/src/ui/components/NumberField.qml:56: Error: Locale: Number.fromLocaleString(): Invalid format
04:17:47 [WARN] qrc:/src/ui/components/NumberField.qml:56: Error: Locale: Number.fromLocaleString(): Invalid format
04:17:47 [WARN] qrc:/src/ui/components/NumberField.qml:56: Error: Locale: Number.fromLocaleString(): Invalid format
04:17:47 [WARN] qrc:/src/ui/components/NumberField.qml:56: Error: Locale: Number.fromLocaleString(): Invalid format
04:17:47 [WARN] qrc:/src/ui/components/NumberField.qml:56: Error: Locale: Number.fromLocaleString(): Invalid format
04:17:47 [WARN] qrc:/src/ui/menu/Export.qml:80:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
04:17:47 [WARN] qrc:/src/ui/menu/Advanced.qml:15:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
04:17:48 [WARN] qrc:/src/ui/components/Timeline.qml:168:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
04:17:48 [WARN] qrc:/src/ui/App.qml:101:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
04:17:48 [INFO] GPU list: ["[OpenCL] NVIDIA CUDA NVIDIA GeForce RTX 3060 Laptop GPU: 3.0", "[OpenCL] AMD Accelerated Parallel Processing gfx1035: 2.0", "[wgpu] NVIDIA GeForce RTX 3060 Laptop GPU (Vulkan)", "[wgpu] AMD Radeon(TM) Graphics (Vulkan)", "[wgpu] NVIDIA GeForce RTX 3060 Laptop GPU (Dx12)", "[wgpu] AMD Radeon(TM) Graphics (Dx12)", "[wgpu] NVIDIA GeForce RTX 3060 Laptop GPU/PCIe/SSE2 (Gl)"]
04:17:48 [WARN] qrc:/src/ui/components/NumberField.qml:56: Error: Locale: Number.fromLocaleString(): Invalid format
[... 64 lines cut ...]
04:17:55 [DEBUG] (20) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11
04:17:55 [DEBUG] (20) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x21483bb3390) QSize(1920, 1080)
04:17:55 [ERROR] thread '<unnamed>' panicked at 'range end index 9223372036854785437 out of range for slice of length 9628': src\core\zooming\zoom_dynamic.rs:117
   0: <unknown>
   1: <unknown>
   2: <unknown>
   3: <unknown>
   4: <unknown>
   5: <unknown>
   6: <unknown>
   7: <unknown>
   8: <unknown>
   9: <unknown>
  10: <unknown>
  11: <unknown>
  12: <unknown>
  13: <unknown>
  14: <unknown>
  15: <unknown>
  16: <unknown>
  17: BaseThreadInitThunk
  18: RtlUserThreadStart

Rayon: detected unexpected panic; aborting
```

AdrianEddy, 2024-07-25:
Download latest dev build https://gyroflow.xyz/devbuild/, open it, go to Advanced on bottom right and Reset all settings to defaults.

If that still crashes, it's because you have wrong locale settings in Windows, go to Settings -> Time and Language -> Language and Region -> Administrative language settings -> Formats -> Additional settings and check your decimal and grouping separators, they can't be the same

[image]

closed 2024-07-25 by AdrianEddy

Yuri-jeik-Becker, 2024-07-25:
Thanks! The issue was the use of comma on both decimal and digit grouping, I changed that to use a brasilian program and forgot ;u;