gyroflow/gyroflow #869: Crash when exporting vertical 4k 4:3

Westpol, 2024-07-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

v1.5.4 gh1646 / v.1.5.4 gh2041

### What operating system are you using?

Linux Mint 21.3

### What GPU are you using?

AMD Radeon RX6800

### What happened?

When I load a 4k 4:3 video and change the aspect ratio so that the video is 4k 4:3 vertical and export it, I get an error message. (link with screenshot of the error message) [https://drive.google.com/file/d/1UatJUWphc0mp--MzEo10uIEFlLmDDbEO/view?usp=sharing](url)

I tried installing the AMD driver, but in that case, the program crashed the whole driver upon opening a video file and I only had a freeze frame until manually rebooting. I also tried different kernel versions, but I still get the same error message.

Thanks in advance for any help :)

### Relevant log output

```shell
Build for: Linux x86_64; Clang17.0.4 (++20231031083134+309d55140c46-1~exp1~20231031083240.62)GCC4.2.1; libc++170004; 10:04:03 Nov  1 2023
22:06:58 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.3
22:06:58 [DEBUG] (1) gyroflow: Gyroflow 1.5.4 (gh1646)
22:06:58 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Clover") Ok("OpenCL 1.1 Mesa 23.2.1-1ubuntu3.1~22.04.2") ["AMD Radeon RX 6800 (navi21, LLVM 15.0.7, DRM 3.57, 6.8.0-39-generic)"]
22:06:58 [INFO] OpenCL Platform: Clover, ext: Extensions { inner: "cl_khr_icd" } Device: AMD AMD Radeon RX 6800 (navi21, LLVM 15.0.7, DRM 3.57, 6.8.0-39-generic)
22:06:58 [DEBUG] (1) gyroflow::rendering: GPU type: Amd, from name: amd amd radeon rx 6800 (navi21, llvm 15.0.7, drm 3.57, 6.8.0-39-generic)
22:06:58 [DEBUG] (1) mdk: [MDK] global option: X11Display = 0x5f2e8a1e1130
22:06:58 [DEBUG] (1) default: [in MDKPlayer::MDKPlayer()] X11 display: 0x5f2e8a1e1130
22:06:59 [INFO] GPU list: ["[OpenCL] Clover AMD Radeon RX 6800 (navi21, LLVM 15.0.7, DRM 3.57, 6.8.0-39-generic): 1.1", "[wgpu] AMD Radeon RX 6800 (RADV NAVI21) (Vulkan)", "[wgpu] llvmpipe (LLVM 15.0.7, 256 bits) (Vulkan)", "[wgpu] AMD Radeon RX 6800 (navi21, LLVM 15.0.7, DRM 3.57, 6.8.0-39-generic) (Gl)"]

22:07:28 [ERROR] OpenCL error init_backends: OclCore(ProgramBuild(BuildLog("fatal error: cannot open file '/usr/lib/clc/gfx1030-amdgcn-mesa-mesa3d.bc': No such file or directory")))
22:07:28 [DEBUG] (2) gyroflow_core::gpu::wgpu: WGPU initializing adapter #0
22:07:28 [INFO] Initialized wgpu for (4096, 3072, 4096) -> (3072, 4096, 3072) | key: 2824608176opencv_fisheye80(4096, 3072)(3072, 4096)
22:07:28 [INFO] Clearing 1 stab data because backend was initialized
22:07:28 [DEBUG] (2) gyroflow_core::stabilization: Setting device 0
22:07:28 [INFO] OpenCL Platform: Clover, Device: AMD AMD Radeon RX 6800 (navi21, LLVM 15.0.7, DRM 3.57, 6.8.0-39-generic)
22:07:28 [ERROR] OpenCL error init_backends: OclCore(ProgramBuild(BuildLog("fatal error: cannot open file '/usr/lib/clc/gfx1030-amdgcn-mesa-mesa3d.bc': No such file or directory")))
22:07:28 [DEBUG] (2) gyroflow_core::gpu::wgpu: WGPU initializing adapter #0
22:07:28 [INFO] Initialized wgpu for (2048, 1536, 4096) -> (1536, 2048, 3072) | key: 3645604960opencv_fisheye80(2048, 1536)(1536, 2048)
22:07:28 [INFO] Clearing 1 stab data because backend was initialized
22:07:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Hardware transfer formats from GPU: Some([NV12, P010LE, YUV420P, YUV420P, YUYV422, UYVY422, GRAY8, YUV444P, BGRA, RGBA, ARGB, BGRZ, RGBZ])
22:07:28 [DEBUG] (2) gyroflow::rendering: No matching codec, we need NV12 and supported are: [VAAPI]

22:07:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type: Some(AV_HWDEVICE_TYPE_VAAPI), encoder_pixel_format: Some(NV12)
22:07:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: NV12, color range: MPEG
22:07:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type Some(AV_HWDEVICE_TYPE_VAAPI)
22:07:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: Setting hw_device_ctx 0x71cf402061c0
22:07:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: target_format: AV_PIX_FMT_NV12
22:07:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: inited hwframe ctx
22:07:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: frames_ctx.format: AV_PIX_FMT_VAAPI
22:07:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: frames_ctx.sw_format: AV_PIX_FMT_NV12
[... 485 lines cut ...]
22:11:39 [DEBUG] (2) gyroflow_core::stabilization: Setting device 0
22:11:39 [INFO] OpenCL Platform: Clover, Device: AMD AMD Radeon RX 6800 (navi21, LLVM 15.0.7, DRM 3.57, 6.8.0-39-generic)
22:11:39 [ERROR] OpenCL error init_backends: OclCore(ProgramBuild(BuildLog("fatal error: cannot open file '/usr/lib/clc/gfx1030-amdgcn-mesa-mesa3d.bc': No such file or directory")))
22:11:39 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type: Some(AV_HWDEVICE_TYPE_VAAPI), encoder_pixel_format: None
22:11:39 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: YUV420P, color range: MPEG
22:11:39 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type Some(AV_HWDEVICE_TYPE_VAAPI)
22:11:39 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: Setting hw_device_ctx 0x71cf40691500
22:11:39 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: target_format: AV_PIX_FMT_YUV420P
22:11:39 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: inited hwframe ctx
22:11:39 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: frames_ctx.format: AV_PIX_FMT_VAAPI
22:11:39 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: frames_ctx.sw_format: AV_PIX_FMT_YUV420P
22:11:39 [WARN] [h264_vaapi @ 0x71cf43724000] Driver does not support some wanted packed headers (wanted 0xd, found 0).
22:11:39 [WARN] [h264_vaapi @ 0x71cf43724000] Driver does not support packed sequence headers, but a global header is requested.
22:11:39 [WARN] [h264_vaapi @ 0x71cf43724000] No global header will be written: this may result in a stream which is not usable for some purposes (e.g. not muxable to some containers).
22:11:39 [ERROR] [h264_vaapi @ 0x71cf43724000] Hardware does not support encoding at size 3072x4096 (constraints: width 256-4096 height 128-2160).

22:11:39 [INFO] OpenCL Platform: Clover, Device: AMD AMD Radeon RX 6800 (navi21, LLVM 15.0.7, DRM 3.57, 6.8.0-39-generic)
22:11:39 [ERROR] OpenCL error init_backends: OclCore(ProgramBuild(BuildLog("fatal error: cannot open file '/usr/lib/clc/gfx1030-amdgcn-mesa-mesa3d.bc': No such file or directory")))

22:11:39 [WARN] [h264_vaapi @ 0x71cf41f8afc0] Driver does not support some wanted packed headers (wanted 0xd, found 0).
22:11:39 [WARN] [h264_vaapi @ 0x71cf41f8afc0] Driver does not support packed sequence headers, but a global header is requested.
22:11:39 [WARN] [h264_vaapi @ 0x71cf41f8afc0] No global header will be written: this may result in a stream which is not usable for some purposes (e.g. not muxable to some containers).
22:11:39 [ERROR] [h264_vaapi @ 0x71cf41f8afc0] Hardware does not support encoding at size 3072x4096 (constraints: width 256-4096 height 128-2160).
22:11:39 [DEBUG] (2) gyroflow::rendering: [aac @ 0x71cf41ce7400] Qavg: nan
```

AdrianEddy, 2024-07-30:
It says in the error message: `Hardware does not support encoding at size 3072x4096 (constraints: width 256-4096 height 128-2160).`
which means that the max resolution you can render on that GPU is 4096x2160

It's a hardware limitation and there's nothing I can do about it. You can still use CPU encoding though.

closed 2024-07-30 by AdrianEddy

AdrianEddy, 2024-07-31:
AMD is really behind NVIDIA when it comes to video encoding and format support. Also, video encoding APIs on Linux are also quite a mess unfortunately

Westpol, 2024-08-01:
Just as a quick info: I just installed Linux Mint 22 which came out a few days ago and Gyroflow works perfectly fine now. So if anyone encounters Gyroflow problems with an AMD GPU and LM21.x, it might be worth it to upgrade to LM22.