gyroflow/gyroflow #872: GPU-Encoding Bitrate does not match on Apple M1 Pro

prafgup, 2024-08-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

v1.5.4

### What operating system are you using?

mcOs Sonoma 14.5 (23F79)

### What happened?

I have the same exact issue with my Apple M1 Pro as #472 , if I disable GPU encoding it works fine and respects my Bitrate that I have set, but gives out a random Bitrate in case its enabled, I have tried setting the `-rc cqp` custom encoder as well as suggested in the comments.

[image]

With GPU - 
[image]

With CPU - 
[image]

AdrianEddy, 2024-08-04:
That was already fixed in the dev build https://gyroflow.xyz/devbuild

closed 2024-08-04 by AdrianEddy

prafgup, 2024-08-04:
Thanks for the update @AdrianEddy 

Seems like the last release https://github.com/gyroflow/gyroflow/releases/tag/v1.5.4 is from `Nov 1, 2023` thats around 10 months old. A new release soon might help others as well :)

AdrianEddy, 2024-08-04:
I'm working on getting 1.6.0 done, but there's still some work to do before release