gyroflow/gyroflow #873: Gyroflow for Mac Crashes on Sony MP4 File

sturmen, 2024-08-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

gh2051

### What operating system are you using?

macOS 14.6.1

### What GPU are you using?

M1 Max

### What happened?

When trying to load the video file, the Gyroflow application crashes.

I have reproduced this on both macOS and Windows.

It appears to be common to all clips I recorded with my Sony a9 III. Here is an example file: https://mega.nz/file/3IpClagI#zGsGmkYFFm87yvFt-Jif1bkm3y7STttEkvb1SMzrdLU

[macOS crashdump.txt](https://github.com/user-attachments/files/16542270/crashdump.txt)

### Relevant log output

```shell
11:31:46 [0m[36m[DEBUG] [0m(1) gyroflow: [in main] Qt version: 6.4.3
11:31:46 [0m[36m[DEBUG] [0m(1) gyroflow: Gyroflow 1.5.4 (gh2051)
11:31:46 [0m[34m[INFO] [0mVideos: ["/Users/nicholastinsley/Desktop/Sony_File_That_Causes_Gyroflow_Crash.MP4"]
11:31:46 [0m[34m[INFO] [0mLens profiles directory: "/Applications/Gyroflow.app/Contents/Resources/camera_presets"
11:31:46 [0m[34m[INFO] [0mLoaded 10164 lens profiles in 126.975ms
11:31:46 [0m[34m[INFO] [0mOpenCL Platform: Apple, ext: Extensions { inner: "cl_APPLE_SetMemObjectDestructor cl_APPLE_ContextLoggingFunctions cl_APPLE_clut cl_APPLE_query_kernel_names cl_APPLE_gl_sharing cl_khr_gl_event" } Device: Apple Apple M1 Max
11:31:46 [0m[34m[INFO] [0mstart_accessing_url: file:///Users/nicholastinsley/Desktop/Sony_File_That_Causes_Gyroflow_Crash.MP4 - OPEN
11:31:46 [0m[33m[WARN] [0m[mov,mp4,m4a,3gp,3g2,mj2 @ 0x143e1bad0] st: 0 edit list: 1 Missing key frame while searching for timestamp: 2002
11:31:46 [0m[33m[WARN] [0m[mov,mp4,m4a,3gp,3g2,mj2 @ 0x143e1bad0] st: 0 edit list 1 Cannot find an index entry before timestamp: 2002.
11:31:46 [0m[33m[WARN] [0m[mov,mp4,m4a,3gp,3g2,mj2 @ 0x143e1bad0] infe version < 2
11:31:46 [0m[33m[WARN] [0mis not implemented. Update your FFmpeg version to the newest one from Git. If the problem still occurs, it means that your file has a feature which has not been implemented.
11:31:46 [0m[33m[WARN] [0m[mov,mp4,m4a,3gp,3g2,mj2 @ 0x143e1bad0] infe version < 2
11:31:46 [0m[33m[WARN] [0mis not implemented. Update your FFmpeg version to the newest one from Git. If the problem still occurs, it means that your file has a feature which has not been implemented.
11:31:46 [0m[34m[INFO] [0mLoaded VideoInfo { duration_ms: 3003.0, frame_count: 180, fps: 59.94005994005994, width: 3840, height: 2160, bitrate: 192.9460735321045, rotation: 0, created_at: Some(1723115983) }
11:31:46 [0m[34m[INFO] [0mCameraIdentifier {
    brand: "Sony",
    model: "ILCE-9M3",
    lens_model: "",
    lens_info: "17.00 mm",
    focal_length: Some(
        17.0,
    ),
    camera_setting: "",
    fps: 59940,
    video_width: 3840,
    video_height: 2160,
    additional: "",
    identifier: "sony-ilce-9m3-17.00mm-3840x2160@59940",
}
0.29.0 (git 5146bc9) - Multimedia Development Kit. Copyright (c) 2016-2024 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: macOS110000/14.6.1 16KB ARMv8 AArch64; libc++170006; Apple Clang15.0.0 (clang-1500.3.9.4); 00:35:07 Aug  7 2024
Crashdump written to /Applications/Gyroflow.app/Contents/MacOS/417AE862-3539-40AD-A1F0-B1B45BE7334F.dmp
```

AdrianEddy, 2024-08-08:
This is because it has global shutter which was not accounted for when we calculate the IBIS data.

Can you send me a sample file with IBIS enabled, and some movement and actual video in there?

AdrianEddy, 2024-08-08:
Latest dev build won't crash, but I'm not sure how well the IBIS will work without a sample file - please test a few different files and let me know if it's all good

sturmen, 2024-08-09:
Thanks for the quick response! I have uploaded a folder of such sample files here: https://mega.nz/folder/CYhGRDjb#XsVZQLdZLyD5cTEFqA0JYQ

Hopefully they cover all of your test cases.

AdrianEddy, 2024-08-09:
Thanks, they are very useful. Not all of them are working properly, but the issues are already tracked in #849 

If I could have one more request, it would be useful to have a sample with "Standard" stabilization and fast sideways panning showing some straight line like a door or window, to mare sure rolling shutter calculations are properly handled in case of global shutter sensor. Thanks!

Since the crash is fixed, I'm closing this issue

closed 2024-08-09 by AdrianEddy

sturmen, 2024-08-13:
@AdrianEddy here is the requested footage: https://mega.nz/file/SVZl0YLT#yOkuOZXmzHPCVg7KBxCIqh_2-fz26ZqbzoFUnG_0xPE

Lines were surprisingly not as straight as I was expecting. I am chalking that up to the IBIS unit not being able to keep up with the movement.