gyroflow/gyroflow #879: Error 0xc0000142 when launching dev build

dizzybro, 2024-08-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4 dev build

### What operating system are you using?

Windows 11

### What GPU are you using?

RTX 3060

### What happened?

When I attempt to launch the program I get an error saying "The application was unable to start correctly (0xc0000142) Click OK to close the application" The standard release of 1.5.4 works fine.

AdrianEddy, 2024-08-15:
Install VC Redist https://aka.ms/vs/17/release/vc_redist.x64.exe

closed 2024-08-15 by AdrianEddy

dizzybro, 2024-08-15:
That did the trick!