gyroflow/gyroflow #881: How to Convert the data of External Gyroscope to Camera's Coordinate System

laocan, 2024-08-21:
I want to externally attach a gyroscope to my camera (rigidly connected, without any relative displacement). How can I convert the gyroscope data into the camera's "IMU" data and pass it to Gyroflow for video stabilization?

AdrianEddy, 2024-08-21:
not sure what you mean? Are you building your own gyro-capture device, or using any of existing ones? The orientation is determined by IMU orientation field in Gyroflow, you should set it accordingly

laocan, 2024-08-22:
[image]
The positional relationship between the camera and the IMU is shown in the image. A 4x4 matrix for IMU2CAM (not a 3x3 matrix) has been calculated through calibration. How should I convert the quaternion from the gyroscope's coordinate system to the corresponding quaternion in the camera's coordinate system? Or, as shown in the code below, can I directly convert it into the final rotation matrix?

https://github.com/gyroflow/gyroflow/blob/e7b7c2f5b37782801f0438fa3b0f434a6c91639a/src/core/stabilization/frame_transform.rs#L219

AdrianEddy, 2024-08-22:
but what exactly do you need this for? Are you writing your own code? are you using `gyroflow-core`? Are you asking about .gcsv? Are you asking just in the Gyroflow's user interface?

The settings controlling this are `IMU orientation` and `IMU rotation` in the UI
https://github.com/gyroflow/gyroflow/blob/master/src/core/gyro_source/imu_transforms.rs#L9-L10