gyroflow/gyroflow #882: Freeze when lunching v1.5.4

ScrappyC0co31, 2024-08-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

v1.5.4

### What operating system are you using?

Windows 11

### What GPU are you using?

NVDIA GEFORCE RTX 3050 Ti

### What happened?

After a long while I decided to stabilize my latest video and tried to lunch Gyroflow v1.5.4 which I downloaded back in May 2024. I noticed it just keeps spinning and cannot initialize the software. It seems frozen and the spinning icon stays. When you try to click anywhere it does nothing and sometimes it says program doesn't respond. I have tried to reinstall via zip file and even on Microsoft Store to now avail. Please help
[image]

AdrianEddy, 2024-08-25:
Already fixed in the dev build https://gyroflow.xyz/devbuild

closed 2024-08-25 by AdrianEddy

ScrappyC0co31, 2024-08-26:
> Already fixed in the dev build https://gyroflow.xyz/devbuild

I'm afraid I have tried dev or nightly build thrice already. 
[image]

ScrappyC0co31, 2024-08-26:
03:49:51 [WARN] qrc:/src/ui/main_window.qml:38:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
03:49:53 [DEBUG] (1) gyroflow_core::filesystem: path_to_url | path:  -> Ok("")
03:49:53 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 12.6.41") ["NVIDIA GeForce RTX 3050 Ti Laptop GPU"]
03:49:53 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("AMD Accelerated Parallel Processing") Ok("OpenCL 2.1 AMD-APP (3180.7)") ["gfx902"]
03:49:53 [INFO] OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_d3d10_sharing cl_khr_d3d10_sharing cl_nv_d3d11_sharing cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_win32 cl_khr_external_memory_win32" } Device: NVIDIA Corporation NVIDIA GeForce RTX 3050 Ti Laptop GPU
03:49:54 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
03:49:54 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: HWDevice::from_type AV_HWDEVICE_TYPE_CUDA, device: None
03:49:54 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
03:49:54 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia corporation nvidia geforce rtx 3050 ti laptop gpu
03:49:56 [WARN] qrc:/src/ui/menu/Advanced.qml:15:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
03:49:56 [WARN] qrc:/src/ui/menu/Synchronization.qml:17:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
03:49:56 [WARN] qrc:/src/ui/menu/Stabilization.qml:25:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
03:49:57 [WARN] qrc:/src/ui/menu/Export.qml:80:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
03:49:57 [WARN] qrc:/src/ui/components/Timeline.qml:234:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
03:49:57 [WARN] qrc:/src/ui/App.qml:101:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.

AdrianEddy, 2024-08-26:
If it also doesn't work in the dev build then download and install latest drivers from nvidia.com AND from AMD.com for your integrated GPU