gyroflow/gyroflow #885: MacOs Gyroflow app can't join files, shows PermissionDenied error

vedmant, 2024-09-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4

### What operating system are you using?

MacOs Sonoma 14.2.1 (23C71)

### What GPU are you using?

M1

### What happened?

When I move multiple files (video split from DJI O3) into Gyroflow and select to merge files, it shows PermissionDenied error.

This works ok with [the latest development build](https://gyroflow.xyz/devbuild/) but doesn't work when app is installed from App Store.

I tried to add Gyroflow to full disk access list of apps, but it didn't help.

AdrianEddy, 2024-09-05:
Try the TestFlight version: https://testflight.apple.com/join/e0RI1n5i

Also, when joining files, Gyroflow asks your for the output location. Are you sure you have write permissions to that directory?

vedmant, 2024-09-06:
@AdrianEddy It's Downloads folder on Mac, the files I drop to Gyroflow are also located there. I'll check TestFlight version.

AdrianEddy, 2024-09-14:
Improved in 73bea1edfc0a03342144b53ae3efbbd8119c9967
TestFlight version 1.54.2102

closed 2024-09-14 by AdrianEddy