gyroflow/gyroflow #891: Ubuntu compile fail in rustrover

Divenire1998, 2024-09-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

9395ebd1cb0598f63da7bea9cbfb5a9da046ebad

### What operating system are you using?

Ubuntu 20.04

### What GPU are you using?

4060ti

### What happened?

I am using Ubuntu20.04 to build whole program, follow the instructions by just install-deps and just run,
the program can run .
howere ,when i using RustRover ide try to debug the program,  build generate errors as below:
[image]
[image]

### Relevant log output

```shell
error: failed to run custom build command for `qml-video-rs v0.1.0 (https://github.com/AdrianEddy/qml-video-rs.git?rev=2a3bd8b#2a3bd8ba)`
note: To improve backtraces for build dependencies, set the CARGO_PROFILE_RELEASE_BUILD_OVERRIDE_DEBUG=true environment variable to enable debug information generation.
Caused by:
  process didn't exit successfully: `/media/qiuyue/2_data/0_Workspace/yvr_ws/Stereo_video_stabilization_ws/opensource_proj/gyroflow/target/release/build/qml-video-rs-17ad0ce874f8db41/build-script-build` (exit status: 1)
  --- stdout
  cargo:rustc-link-search=/media/qiuyue/2_data/0_Workspace/yvr_ws/Stereo_video_stabilization_ws/opensource_proj/gyroflow/target/release/build/qml-video-rs-07ca0e6759dd6606/out/mdk-sdk/lib/amd64/
  cargo:rustc-link-lib=mdk
  OUT_DIR = Some(/media/qiuyue/2_data/0_Workspace/yvr_ws/Stereo_video_stabilization_ws/opensource_proj/gyroflow/target/release/build/qml-video-rs-07ca0e6759dd6606/out)
  TARGET = Some(x86_64-unknown-linux-gnu)
  OPT_LEVEL = Some(3)
  HOST = Some(x86_64-unknown-linux-gnu)
  cargo:rerun-if-env-changed=CXX_x86_64-unknown-linux-gnu
  CXX_x86_64-unknown-linux-gnu = None
  cargo:rerun-if-env-changed=CXX_x86_64_unknown_linux_gnu
  CXX_x86_64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=HOST_CXX
  HOST_CXX = None
  cargo:rerun-if-env-changed=CXX
  CXX = None
  cargo:rerun-if-env-changed=CC_ENABLE_DEBUG_OUTPUT
  RUSTC_WRAPPER = None
  cargo:rerun-if-env-changed=CRATE_CC_NO_DEFAULTS
  CRATE_CC_NO_DEFAULTS = None
  DEBUG = Some(false)
  CARGO_CFG_TARGET_FEATURE = Some(fxsr,sse,sse2)
  cargo:rerun-if-env-changed=CXXFLAGS_x86_64-unknown-linux-gnu
  CXXFLAGS_x86_64-unknown-linux-gnu = None
  cargo:rerun-if-env-changed=CXXFLAGS_x86_64_unknown_linux_gnu
  CXXFLAGS_x86_64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=HOST_CXXFLAGS
  HOST_CXXFLAGS = None
  cargo:rerun-if-env-changed=CXXFLAGS
  CXXFLAGS = None
  OUT_DIR = Some(/media/qiuyue/2_data/0_Workspace/yvr_ws/Stereo_video_stabilization_ws/opensource_proj/gyroflow/target/release/build/qml-video-rs-07ca0e6759dd6606/out)
  cargo:rerun-if-env-changed=CC_ENABLE_DEBUG_OUTPUT
  cargo:rerun-if-env-changed=CRATE_CC_NO_DEFAULTS
  CRATE_CC_NO_DEFAULTS = None
  CARGO_CFG_TARGET_FEATURE = Some(fxsr,sse,sse2)
  cargo:rerun-if-env-changed=CXXFLAGS_x86_64-unknown-linux-gnu
  CXXFLAGS_x86_64-unknown-linux-gnu = None
  cargo:rerun-if-env-changed=CXXFLAGS_x86_64_unknown_linux_gnu
  CXXFLAGS_x86_64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=HOST_CXXFLAGS
  HOST_CXXFLAGS = None
  cargo:rerun-if-env-changed=CXXFLAGS
  CXXFLAGS = None
  cargo:warning=In file included from src/video_player.rs:6:
  cargo:warning=/home/qiuyue/.cargo/git/checkouts/qml-video-rs-a29d2c8701d704e5/2a3bd8b/src/cpp/VideoTextureNode.h:6:10: fatal error: private/qquickitem_p.h: No such file or directory
  cargo:warning=    6 | #include <private/qquickitem_p.h>
  cargo:warning=      |          ^~~~~~~~~~~~~~~~~~~~~~~~
  cargo:warning=compilation terminated.
  --- stderr
  error occurred: ToolExecError: Command "c++" "-O3" "-ffunction-sections" "-fdata-sections" "-fPIC" "-m64" "-I" "/home/qiuyue/.cargo/git/checkouts/qml-video-rs-a29d2c8701d704e5/2a3bd8b" "-I" "/usr/include/x86_64-linux-gnu/qt5/QtCore" "-I" "/usr/include/x86_64-linux-gnu/qt5/QtGui" "-I" "/usr/include/x86_64-linux-gnu/qt5/QtQuick" "-I" "/usr/include/x86_64-linux-gnu/qt5/QtQml" "-I" "/usr/include/x86_64-linux-gnu/qt5/QtCore/5.12.8" "-I" "/usr/include/x86_64-linux-gnu/qt5/QtCore/5.12.8/QtCore" "-I" "/usr/include/x86_64-linux-gnu/qt5/QtGui/5.12.8" "-I" "/usr/include/x86_64-linux-gnu/qt5/QtGui/5.12.8/QtGui" "-I" "/usr/include/x86_64-linux-gnu/qt5/QtQuick/5.12.8" "-I" "/usr/include/x86_64-linux-gnu/qt5/QtQuick/5.12.8/QtQuick" "-I" "/usr/include/x86_64-linux-gnu/qt5/QtQml/5.12.8" "-I" "/usr/include/x86_64-linux-gnu/qt5/QtQml/5.12.8/QtQml" "-I" "/media/qiuyue/2_data/0_Workspace/yvr_ws/Stereo_video_stabilization_ws/opensource_proj/gyroflow/target/release/build/qml-video-rs-07ca0e6759dd6606/out/mdk-sdk/include/" "-I" "/usr/include/x86_64-linux-gnu/qt5" "-Wall" "-Wextra" "-std=c++11" "-o" "/media/qiuyue/2_data/0_Workspace/yvr_ws/Stereo_video_stabilization_ws/opensource_proj/gyroflow/target/release/build/qml-video-rs-07ca0e6759dd6606/out/26a388e6a19cf56d-cpp_closures.o" "-c" "/media/qiuyue/2_data/0_Workspace/yvr_ws/Stereo_video_stabilization_ws/opensource_proj/gyroflow/target/release/build/qml-video-rs-07ca0e6759dd6606/out/rust_cpp/cpp_closures.cpp" with args c++ did not execute successfully (status code exit status: 1).
Process finished with exit code 130 (interrupted by signal 2:SIGINT)

error: failed to run custom build command for `gyroflow v1.5.4 (/media/qiuyue/2_data/0_Workspace/yvr_ws/Stereo_video_stabilization_ws/opensource_proj/gyroflow)`
note: To improve backtraces for build dependencies, set the CARGO_PROFILE_RELEASE_BUILD_OVERRIDE_DEBUG=true environment variable to enable debug information generation.
Caused by:
  process didn't exit successfully: `/media/qiuyue/2_data/0_Workspace/yvr_ws/Stereo_video_stabilization_ws/opensource_proj/gyroflow/target/release/build/gyroflow-16950f670ec6559a/build-script-build` (exit status: 101)
  --- stdout
  cargo::rustc-check-cfg=cfg(compiled_qml)
  cargo:rerun-if-changed=src/qt_gpu/qrhi_undistort.cpp
  --- stderr
  thread 'main' panicked at build.rs:205:87:
  called `Result::unwrap()` on an `Err` value: NotPresent
  stack backtrace:
     0: rust_begin_unwind
     1: core::panicking::panic_fmt
     2: core::result::unwrap_failed
     3: build_script_build::main
     4: core::ops::function::FnOnce::call_once
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

AdrianEddy, 2024-09-12:
If you're not using the `just` scripts then you have to set the environment variables for all the dependencies manually, which is quite complex. I'm not using RustRover myself so I don't know how to set that up there. Just check what the just scripts set and figure out how to set the same variables in your IDE

closed 2024-09-12 by AdrianEddy

Divenire1998, 2024-09-13:
thanks for your  reply , i will try it.