gyroflow/gyroflow #893: Allow to export audio if speed is changed

dany123, 2024-09-14:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

I can see that currently this is intentionately disabled but I'm speeding up a clip to 10x and the preview in Gyroflow has audio and adds a bit of context to the video: there's low background wind and a "whoosh" every time a car passes by which makes the sped up video so much better. I'd like to be able to have this in the final export as it saves time later in the editing workflow.

AdrianEddy, 2024-09-14:
This is out of scope - the amount of work required to process audio is quite significant and the use case is very niche. I don't have the capacity to work on this

If you need this, use the Adobe or OpenFX plugin

closed 2024-09-14 by AdrianEddy

dany123, 2024-09-14:
Oh I assumed that if audio works in preview it was just an artificial limitation. I'll use the video editor then.