gyroflow/gyroflow #894: Braw Files over 32gb crash the app and don't stabilize

Yamboss, 2024-09-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.1-1.5.4

### What operating system are you using?

os ventura 13.6.9

### What GPU are you using?

m1

### What happened?

okay first of all i want to thank you all for this app it's incredible !!!
i use it for my gopro and my dji O3 fpv footage and it's working everytime like a charm !

the issue i'm bringing up is that when i try to load Braw files over 32gb.
when i open the file it just get's stuck and shutdown the app,
in 1.5.1 it's trying to open the Motion data then gets stuck...

itried to use the plugin in davinci instead but i can't fegure it out how to load presets and how to make them if i can't load the file in the gyroflow app becouse it crashes before i can do that...

thank you and im sorry if i missed anything. and it's just a simple answer.

anyways my next move is to try to split the braw without hurting the gyro data somewho

AdrianEddy, 2024-09-17:
Have you tried the dev build? https://gyroflow.xyz/devbuild

Yamboss, 2024-09-17:
i right on it ! will let you know thank you so much for helping out !

Yamboss, 2024-09-17:
same thing on the dev build.. im getting a shutdown and this report to send to apple...

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               gyroflow [11221]
Path:                  /Applications/Gyroflow_DEV.app/Contents/MacOS/gyroflow
Identifier:            xyz.gyroflow
Version:               1.5.4 (1.5.4)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-09-17 12:40:20.9362 +0300
OS Version:            macOS 13.6.9 (22G830)
Report Version:        12
Anonymous UUID:        568286A3-EA48-E9B9-B2A6-4EDD7A89C6A9

Sleep/Wake UUID:       CBB88F08-5A4F-491A-A2EF-3656341A5FD7

Time Awake Since Boot: 37000 seconds
Time Since Wake:       6094 seconds

System Integrity Protection: enabled

Crashed Thread:        11

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   gyroflow [11221]

Application Specific Information:
abort() called

Kernel Triage:
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage

Thread 0:: main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x1a605b6f0 __psynch_cvwait + 8
1   libsystem_pthread.dylib       	       0x1a6098574 _pthread_cond_wait + 1232
2   QtCore                        	       0x109622878 0x109448000 + 1943672
3   QtCore                        	       0x1096227cc QWaitCondition::wait(QMutex*, QDeadlineTimer) + 108
4   QtQuick                       	       0x10ad67f30 0x10ab00000 + 2522928
5   QtQuick                       	       0x10ad6838c 0x10ab00000 + 2524044
6   QtQuick                       	       0x10abe891c QQuickWindow::event(QEvent*) + 3828
7   QtCore                        	       0x1094b131c QCoreApplicationPrivate::notify_helper(QObject*, QEvent*) + 384
8   QtCore                        	       0x1094b0e9c QCoreApplication::notifyInternal2(QObject*, QEvent*) + 292
9   QtGui                         	       0x108a0c4c0 QPlatformWindow::deliverUpdateRequest() + 68
10  libqcocoa.dylib               	       0x10936c4a0 0x109328000 + 279712
11  libqcocoa.dylib               	       0x109358a0c 0x109328000 + 199180
12  libdispatch.dylib             	       0x1a5ee8400 _dispatch_client_callout + 20
13  libdispatch.dylib             	       0x1a5eeb884 _dispatch_continuation_pop + 504
14  libdispatch.dylib             	       0x1a5efee7c _dispatch_source_invoke + 1588
15  libdispatch.dylib             	       0x1a5ef6b4c _dispatch_main_queue_drain + 756
16  libdispatch.dylib             	       0x1a5ef6848 _dispatch_main_queue_callback_4CF + 44
17  CoreFoundation                	       0x1a61b7b84 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 16
18  CoreFoundation                	       0x1a6175304 __CFRunLoopRun + 1992
19  CoreFoundation                	       0x1a61743e8 CFRunLoopRunSpecific + 612
20  HIToolbox                     	       0x1af9cadf0 RunCurrentEventLoopInMode + 292
21  HIToolbox                     	       0x1af9cac2c ReceiveNextEventCommon + 648
22  HIToolbox                     	       0x1af9ca984 _BlockUntilNextEventMatchingListInModeWithFilter + 76
23  AppKit                        	       0x1a939c134 _DPSNextEvent + 636
24  AppKit                        	       0x1a939b2d0 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 716
25  AppKit                        	       0x1a938f734 -[NSApplication run] + 464
26  libqcocoa.dylib               	       0x10933d010 0x109328000 + 86032
[... 1467 lines cut ...]
      "experimentId" : "6643969b3099cf28e049862f",
      "deploymentId" : 400000007
    }
  ]
}
}

Model: Macmini9,1, BootROM 10151.140.19, proc 8:4:4 processors, 8 GB, SMC 
Graphics: Apple M1, Apple M1, Built-In
Display: LG HDR 4K, 5120 x 2880 (5K/UHD+ - Ultra High Definition Plus), Main, MirrorOff, Online
Memory Module: LPDDR4, Micron
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4378), wl0: May 13 2023 07:20:48 version 18.20.383.15.7.8.150 FWID 01-b37727a5
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Ethernet, Ethernet, en0
USB Device: USB30Bus
USB Device: USB31Bus
USB Device: USB3.0 Hub
USB Device: USB2.0 Hub
USB Device: USB 2.0 Hub
USB Device: USB 10/100 LAN
USB Device: USB Billboard Device
USB Device: USB31Bus
USB Device: ASM236X series
Thunderbolt Bus: Mac mini, Apple Inc.
Thunderbolt Bus: Mac mini, Apple Inc.

Yamboss, 2024-09-17:
when loading a 29gb file it's working flowless... is it a harddrive thing ?

AdrianEddy, 2024-09-17:
I will need that braw file

AdrianEddy, 2024-09-17:
I don't think size is the problem because I have a 51 GB BRAW file which works fine. It's probably something in that one particular file, or it's damaged or something. In any case, I'll need that file. You can use https://www.swisstransfer.com/

Yamboss, 2024-09-17:
Will get home later today and send it to you it might be corrupted …Sent from my iPhone

AdrianEddy, 2024-09-25:
were you able to upload that braw file?

Yamboss, 2024-09-25:
No sorry I can’t find a way to send it for free But I’ve checked and you were right The gyro data is missing or corrupt Sent from my iPhone

AdrianEddy, 2024-09-25:
You can send it via https://www.swisstransfer.com/
It's possible that I would be able to recover the gyro data, but I'll need that file to check

Yamboss, 2024-09-26:
sending it - please don't share it :)

Yamboss, 2024-09-26:
i just thought of that' maybe you can fix the corrupt files automatically,
 in gyroflow and only then load the video up like vlc do, just make it to
fix the gyro as well...

Just an idea. I don't know how complex it is, not a code guy :)
Thank you so much for being helpful ! You are the best !

Yamboss, 2024-09-26:
https://www.swisstransfer.com/d/b9260d2b-0246-4be5-aadf-6e717f998356
download link hope you will get it and it can help the gyroflow project !
thanks again

AdrianEddy, 2024-09-26:
I was able to fix the parsing and recover the gyro data. Thanks for the file

 I'll let you know when the updated Gyroflow version is available, probably tomorrow or the day after tomorrow

closed 2024-09-26 by AdrianEddy

Yamboss, 2024-09-26:
Awesome, thanks for the update. you are really really awesome !

AdrianEddy, 2024-09-28:
It's fixed in the latest dev build https://gyroflow.xyz/devbuild