gyroflow/gyroflow #895: Program crashes after pressing create new in lens profile

Bigfrishh, 2024-09-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4

### What operating system are you using?

Win 10 22H2

### What GPU are you using?

EVGA GeForce GTX 1080 Ti

### What happened?

In lens profile, when clicking the create new button, the create new tab appears. Shortly (say 1 second) after however, the whole program closed itself

AdrianEddy, 2024-09-22:
First, have you tried the dev build from https://gyroflow.xyz/devbuild/
Second, do you have latest GPU drivers? and what's your CPU?

Bigfrishh, 2024-09-22:
Intel i9-9900K
Latest GPU drivers confirmed

Issue does not appear in dev build

AdrianEddy, 2024-09-22:
ok then it's already fixed in the dev build

closed 2024-09-22 by AdrianEddy

Bigfrishh, 2024-09-22:
Thank you for your quick response!