gyroflow/gyroflow #897: Support Kandao Qoocam EGO 3D camera

renaudguerin, 2024-09-28:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

The [Qoocam EGO by Kandao](https://www.kandaovr.com/support/qoocam-ego) is a consumer 3D camera that embeds gyro metadata in its video files.
Stabilization is available in proprietary [post-processing apps](https://www.kandaovr.com/support/qoocam-ego#support_detail_download_software) and even [in camera](https://us.kandaovr.com/blogs/news/how-to-stabilize-videos-shot-by-qoocam-ego-in-camera-stabilization), but they offer limited flexibility.

It would be amazing if this could be supported in Gyroflow / telemetry-parser ! I've confirmed that this is not currently the case : 
```
thread 'main' panicked at gyro2bb.rs:38:113:
called `Result::unwrap()` on an `Err` value: Custom { kind: Other, error: "Unsupported file format" }```

Happy to share a sample video file if this is of interest.

Thanks

AdrianEddy, 2024-09-29:
I would need a sample video file and the gyro file is it's separate

renaudguerin, 2024-09-30:
Thanks ! [Here you go](https://www.dropbox.com/scl/fi/s1cdvdqyd7w6gw8gqm8sr/qoocam_ego_sample.mp4?rlkey=r7021agxjt1xuxgk3d6mkf4cu&dl=1)

There is no separate gyro file. If needed I can also provide the same video after in-camera stabilization

AdrianEddy, 2024-10-08:
I implemented the parser in https://github.com/AdrianEddy/telemetry-parser/commit/a4ad182c52b75bcf6cb6ba669c96ed608937356f but I would need a different video (where you can actually see the motion) to determine proper IMU orientation

In any case, 3D video won't be supported in Gyroflow, as Gyroflow is not designed to work with 3D or 360 videos. You can export the gyro data though and do something with it in other software, but plug-and-play support of these videos is not in scope of Gyroflow

closed 2024-10-08 by AdrianEddy

renaudguerin, 2024-10-13:
> I implemented the parser in [AdrianEddy/telemetry-parser@a4ad182](https://github.com/AdrianEddy/telemetry-parser/commit/a4ad182c52b75bcf6cb6ba669c96ed608937356f) but I would need a different video (where you can actually see the motion) to determine proper IMU orientation

Thanks ! [Here is one](https://www.dropbox.com/scl/fi/at1nz3gza32c2w9m8wg4k/0024_20241013_155430_01.mp4?rlkey=vir2yhn7fz85t2ol3uabx6jpm&dl=0)

> In any case, 3D video won't be supported in Gyroflow, as Gyroflow is not designed to work with 3D or 360 videos. You can export the gyro data though and do something with it in other software, but plug-and-play support of these videos is not in scope of Gyroflow

I don't know of any other software that could make use of the gyro data, it seems that you did a great job and Gyroflow is the standard in this space :)

If 3D video is out of scope, I suppose my best bet is to split the original 3D SBS file into separate videos for each eye, somehow preserving the gyro metadata in each, then process them one by one in Gyroflow then reassemble the results into 3D SBS ?

Do you think that should work ? Or is there a risk Gyroflow might process the two eye videos inconsistently in spite of identical gyro data, making them impossible to reassemble ?

AdrianEddy, 2024-10-13:
This video is still not enough, there's a very small motion in one axis - it's not enough to determine the correct movement. Also, you'll need the lens profile, so just go ahead and film the calibration pattern and make sure to rotate the camera in all 3 axes
https://docs.gyroflow.xyz/app/getting-started/lens-calibration

After that, you'll be able to do like you said, separate the sides to two separate videos, stabilize in gyroflow and then join again. It's possible to automate this to some extent (or maybe even fully) using ffmpeg and gyroflow cli

renaudguerin, 2024-11-05:
Here is a [calibration video](https://www.icloud.com/iclouddrive/007zrhih_fnJMQa-cNBRqP0-g#koocam_ego_calibration)

Hopefully this one has data on all axes. Can I use it as is to create a lens profile, or would I need to separate the sides first and use just one of them for calibration ?

AdrianEddy, 2024-11-05:
You have to calibrate only one lens

Here's the calibrated profile for one lens: 
[KanDaoEgo-left.json](https://github.com/user-attachments/files/17637104/KanDaoEgo-left.json)

Sync offset is around -950ms
IMU Orientation is **XYZ**