gyroflow/gyroflow #900: merge gcsv fail, gyroflow v1.54 and dev 1.54(gh2148)

robin-debug, 2024-10-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

v1.54 and dev 1.54(gh2148)

### What operating system are you using?

Windows 11

### What happened?

merge 3 gcsv file into one,the joined file has bad time length,maybe bad time index.
[gyroflow.log](https://github.com/user-attachments/files/17269131/gyroflow.log)
[gcsv.zip](https://github.com/user-attachments/files/17269133/gcsv.zip)
[image]

robin-debug, 2024-10-06:
I guess the bug in src/controller.rs line 2172,in function merge_gcsv
[image]
the 1st gcsv have 11947 timestamps,and the 2nd have 11951,
when run in 2nd file,work with 11948 timestamp,
base on code 2172,the add_timestamp reset to 0
then all left timestamp is bad
[image]
and because 3th gcsv have more timestamp(11967) then 2nd(11951),same bug again
[image]

AdrianEddy, 2024-10-06:
Fixed in 8a989f7be0aa29cf839f0085e250cd413bb0462d

closed 2024-10-06 by AdrianEddy