gyroflow/gyroflow #901: Lanczos4 interpolation compresses full color range into limited

liskdev, 2024-10-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

Dev build downloaded on 7 Oct 2024

### What operating system are you using?

Linux

### What GPU are you using?

NVIDIA GeForce RTX 4080 Laptop GPU

### What happened?

Comparing stable 1.5.4 version with dev build i found that default Lanczoz4 interpolation compresses original video full (JPEG) range into limited (tv).

- Input: Video Gopro11 10bit 5k video with full range.
- Settings: No Stabilization, original video size (copy 1 to 1, basically no changes in video)
- Compression: hevc software. 
- Result: It saves file with full color range metadata flag but i fact it is compressed to tv range. (Verified by histogram in Davinci)

When i use new RobidouxSharp interpolation then all good.

I could provide file examples if need later.

Thanks.

AdrianEddy, 2024-10-09:
huh, that's very weird, the interpolation method doesn't touch the color values. Are you absolutely sure that the only difference was the interpolation method? Also which, video encoder was used, GPU-accelerated NVIDIA?
Sample files will be useful

AdrianEddy, 2024-10-09:
One more question, does it behave the same if you change "Device for video processing" between [OpenCL] and [wgpu]?

AdrianEddy, 2024-10-09:
can you check how it behaves before the interpolation method changes, in gh2150? 
https://github.com/gyroflow/gyroflow/actions/runs/11202849808/artifacts/2020871827

AdrianEddy, 2024-10-09:
Also in gh2158 https://github.com/gyroflow/gyroflow/actions/runs/11254703273/artifacts/2034758315

liskdev, 2024-10-09:
gh2150 has the same issue. I guess there is no sense to test gh2158?

liskdev, 2024-10-09:
I made short 1sec cut for tests, it lacks gyro information so app complains about it, but it shoudl work with No Smoothing option.

Here are files https://drive.google.com/drive/folders/1Om6c3-vCdBsxYxEtSsG-Kn6ebf-8GdrQ?usp=sharing

- I tested with old 1.5.4 release (gh1646) - all good in all cases.
- Then gh2150, gh2151 (this i had initially) and gh2158. 
- They all have an issue if compression is NOT GPU. With GPU compression they all fine. 
- Setting video processing to wgpu, wgpu(Vulcan), opencl has no effect.
- Changing scaling between lanczos and new one also doesnt help with this particular 1sec/nogyro file :shrug: , I will double check with other files since im sure there was an effect.

Regarding files: the bad effect looks like image loosing contrast and shadows become lighter.

Let me know what else i can do.

liskdev, 2024-10-09:
Additional detail: it also works well if both decoding and encoding are NOT GPU.

So it seems there is an issue with gpu and cpu 10bit color format conversion.

liskdev, 2024-10-09:
I'm sorry, it seems it was false alarm about scaling algorithm. Then only option which affects result is `Use GPU encoding`
If it is different from decoding then color range is screwed up.
 
I reviewed logs and it seems it was like:
- I rendered couple videos and opened them in separate players to compare
- I tried new scaling option in app
- GPU decoder failed due to GPU memory was used by multiple video players
- app switched to CPU decoder
- and produced correct result since both decoder and encoder were on CPU

AdrianEddy, 2024-10-09:
try gh2159 https://github.com/gyroflow/gyroflow/actions/runs/11261764620/artifacts/2036890775

liskdev, 2024-10-10:
This one works well, i have try as much as possible combinations of processing device, decoder, encoder, interpolation and all seems fine now. Thank you!

btw, when app is opened with wgpu-Gl processing, all works well but there are errors in logs:
```
10:10:16 [DEBUG] (2) gyroflow::rendering::ffmpeg_processor: Codec formats: [YUV420P, YUVJ420P, YUV422P, YUVJ422P, YUV444P, YUVJ444P, GBRP, YUV420P10LE, YUV422P10LE, YUV444P10LE, GBRP10LE, YUV420P12LE, YUV422P12LE, YUV444P12LE, GBRP12LE, GRAY8, GRAY10LE, GRAY12LE]
10:10:16 [DEBUG] (2) gyroflow::rendering::ffmpeg_processor: Output metadata: {"major_brand": "mp41", "minor_version": "538120216", "compatible_brands": "mp41", "timecode": "14:23:43:22", "firmware": "H22.01.01.12.00", "comment": "Original filename: 20240617_142435.mp4", "creation_time": "2024-06-17T12:24:36.989+00:00"}
10:10:16 [DEBUG] (2) gyroflow_core::stabilization: Setting device 4
10:10:16 [DEBUG] (2) gyroflow_core::gpu::wgpu: WGPU adapter (3): AdapterInfo { name: "NVIDIA GeForce RTX 4080 Laptop GPU/PCIe/SSE2", vendor: 4318, device: 0, device_type: Other, driver: "", driver_info: "3.3.0 NVIDIA 550.120", backend: Gl }
10:10:16 [DEBUG] (2) gyroflow_core::gpu::wgpu: WGPU initializing adapter #3: AdapterInfo { name: "NVIDIA GeForce RTX 4080 Laptop GPU/PCIe/SSE2", vendor: 4318, device: 0, device_type: Other, driver: "", driver_info: "3.3.0 NVIDIA 550.120", backend: Gl }
10:10:16 [ERROR] thread '<unnamed>' panicked at 'called `Option::unwrap()` on a `None` value': /root/.cargo/registry/src/index.crates.io-6f17d22bba15001f/khronos-egl-6.0.0/src/lib.rs:1095
   0: <unknown>
...
  29: <unknown>

10:10:16 [ERROR] Failed to initialize wgpu called `Option::unwrap()` on a `None` value
10:10:16 [ERROR] No cached wgpu found for key: 3267390781|opencv_fisheye||8|0|(5312, 4648)|(5312, 4648)|Lanczos4|ThreadId(2)
10:10:17 [INFO] Settings saved to "/home/lk/.local/share/Gyroflow/settings.json"
10:10:17 [DEBUG] (2) gyroflow_core::stabilization: Setting device 4
10:10:17 [DEBUG] (2) gyroflow_core::gpu::wgpu: WGPU adapter (3): AdapterInfo { name: "NVIDIA GeForce RTX 4080 Laptop GPU/PCIe/SSE2", vendor: 4318, device: 0, device_type: Other, driver: "", driver_info: "3.3.0 NVIDIA 550.120", backend: Gl }
10:10:17 [DEBUG] (2) gyroflow_core::gpu::wgpu: WGPU initializing adapter #3: AdapterInfo { name: "NVIDIA GeForce RTX 4080 Laptop GPU/PCIe/SSE2", vendor: 4318, device: 0, device_type: Other, driver: "", driver_info: "3.3.0 NVIDIA 550.120", backend: Gl }
10:10:17 [ERROR] thread '<unnamed>' panicked at 'called `Option::unwrap()` on a `None` value': /root/.cargo/registry/src/index.crates.io-6f17d22bba15001f/khronos-egl-6.0.0/src/lib.rs:1095
   0: <unknown>
...
  28: <unknown>

10:10:17 [ERROR] Failed to initialize wgpu called `Option::unwrap()` on a `None` value
10:10:17 [ERROR] No cached wgpu found for key: 3734948344|opencv_fisheye||8|0|(5312, 4648)|(5312, 4648)|Lanczos4|ThreadId(2)
10:10:17 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Hardware transfer formats from GPU: Some([P010LE])
10:10:17 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Converting from P010LE to YUV420P10LE
10:10:17 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type: None, encoder_pixel_format: Some(YUV420P10LE)
10:10:17 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: YUV420P10LE, color range: JPEG
10:10:17 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type None
x265 [info]: HEVC encoder version 3.6+1-ede03d96d
x265 [info]: build info [Linux][clang 18.1.8][64 bit] 10bit
x265 [info]: using cpu capabilities: MMX2 SSE2Fast LZCNT SSSE3 SSE4.2 AVX FMA3 BMI2 AVX2
x265 [info]: Main 10 profile, Level-6 (High tier)
x265 [info]: Thread pool created using 32 threads
x265 [info]: Slices                              : 1
x265 [info]: frame threads / pool features       : 6 / wpp(73 rows)
x265 [warning]: VBV maxrate specified, but no bufsize, ignored
x265 [info]: Coding QT: max CU size, min CU size : 64 / 8
x265 [info]: Residual QT: max TU size, max depth : 32 / 1 inter / 1 intra
x265 [info]: ME / range / subpel / merge         : hex / 57 / 2 / 3
x265 [info]: Keyframe min / max / scenecut / bias  : 5 / 59 / 40 / 5.00 
x265 [info]: Lookahead / bframes / badapt        : 20 / 4 / 2
x265 [info]: b-pyramid / weightp / weightb       : 1 / 1 / 0
x265 [info]: References / ref-limit  cu / depth  : 3 / off / on
x265 [info]: AQ: mode / str / qg-size / cu-tree  : 2 / 1.0 / 32 / 1
x265 [info]: Rate Control / qCompress            : ABR-94371 kbps / 0.60
x265 [info]: tools: rd=3 psy-rd=2.00 early-skip rskip mode=1 signhide tmvp
x265 [info]: tools: b-intra strong-intra-smoothing lslices=8 deblock sao
10:10:17 [DEBUG] (2) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 12.4.131") ["NVIDIA GeForce RTX 4080 Laptop GPU"]
10:10:17 [INFO] OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_opaque_fd cl_khr_external_memory_opaque_fd" } Device: NVIDIA Corporation NVIDIA GeForce RTX 4080 Laptop G [... cut ...]
10:10:26 [DEBUG] (2) gyroflow_core::filesystem: FfmpegPathWrapper::drop file:///home/some.mp4
x265 [info]: frame I:      1, Avg QP:15.51  kb/s: 600404.00
x265 [info]: frame P:      7, Avg QP:31.22  kb/s: 204330.29
x265 [info]: frame B:     10, Avg QP:33.42  kb/s: 66162.15
x265 [info]: Weighted P-Frames: Y:0.0% UV:0.0%

encoded 18 frames in 8.93s (2.02 fps), 149574.32 kb/s, Avg QP:31.57
```

It seems it fallbacks to OpenCl in this case. OpenCL and Vulcan produce no errors.
And if I switch to Vulcan/OpenCL and then back to Gl it stops producing these errors.

AdrianEddy, 2024-10-10:
Thanks for testing 🎉

As for the Gl error, there's not much I can do about it. That's why we have 3 implementations of the same code for different APIs because there's just too many platforms and combinations and not everything is compatible with each other

closed 2024-10-10 by AdrianEddy