gyroflow/gyroflow #902: CLI output in dev build as of 2024-10-09 contains a lot of noise

jameslin123, 2024-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4 (gh2164)

### What operating system are you using?

Ubuntu 24.04 LTS

### What GPU are you using?

Nvidia GP107GL [Quadro P400]

### What happened?

In CLI mode, the logs contain a huge amount of noise made up of numbers, apparently for each frame of video. Sample logs are attached below, I had to kill the process after a few seconds, otherwise the log file would become too large.

[gyroflow-cli-output.txt](https://github.com/user-attachments/files/17336649/gyroflow-cli-output.txt)

### Relevant log output

```shell
01:58:09 [0m[36m[DEBUG] [0m(1) gyroflow: [in main] Qt version: 6.4.3

01:58:09 [0m[36m[DEBUG] [0m(1) gyroflow: Gyroflow 1.5.4 (gh2164)

01:58:09 [0m[34m[INFO] [0mVideos: ["/home/james/Videos/A7C2/20241009/20241009_C0400.MP4", "/home/james/Videos/A7C2/20241009/20241009_C0401.MP4", "/home/james/Videos/A7C2/20241009/20241009_C0402.MP4", "/home/james/Videos/A7C2/20241009/20241009_C0403.MP4", "/home/james/Videos/A7C2/20241009/20241009_C0404.MP4", "/home/james/Videos/A7C2/20241009/20241009_C0405.MP4", "/home/james/Videos/A7C2/20241009/20241009_C0406.MP4", "/home/james/Videos/A7C2/20241009/20241009_C0407.MP4", "/home/james/Videos/A7C2/20241009/20241009_C0408.MP4", "/home/james/Videos/A7C2/20241009/20241009_C0409.MP4", "/home/james/Videos/A7C2/20241009/20241009_C0410.MP4", "/home/james/Videos/A7C2/20241009/20241009_C0411.MP4", "/home/james/Videos/A7C2/20241009/20241009_C0412.MP4", "/home/james/Videos/A7C2/20241009/20241009_C0413.MP4"]

01:58:09 [0m[34m[INFO] [0mPresets: ["{output:{bitrate:23,codec:H.265/HEVC,},stabilization:{smoothing_params:[{name:smoothness,value:0.25},{name:smoothness_pitch,value:0.5},{name:smoothness_yaw,value:0.5},{name:smoothness_roll,value:0.5},{name:per_axis,value:0},{name:trim_range_only,value:0},{name:max_smoothness,value:1},{name:alpha_0_1s,value:0.1}],},version:2}"]

01:58:09 [0m[34m[INFO] [0mLens profiles directory: "/tmp/.mount_GyrofljgdHtm/camera_presets"

01:58:10 [0m[34m[INFO] [0mLoaded 12172 lens profiles in 215.124ms

01:58:10 [0m[34m[INFO] [0mOpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_copy_opts cl_khr_gl_event cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_nv_kernel_attribute cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_opaque_fd cl_khr_external_memory_opaque_fd" } Device: NVIDIA Corporation Quadro P400

01:58:10 [0m[34m[INFO] [0mSettings file path: /home/james/.local/share/Gyroflow/settings.json

[mov,mp4,m4a,3gp,3g2,mj2 @ 0x702d28000d00] st: 0 edit list: 1 Missing key frame while searching for timestamp: 1001
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x702d3c000d00] st: 0 edit list: 1 Missing key frame while searching for timestamp: 1001
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x702d24000d00] st: 0 edit list: 1 Missing key frame while searching for timestamp: 1001
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x702d28000d00] st: 0 edit list 1 Cannot find an index entry before timestamp: 1001.

01:58:10 [0m[33m[WARN] [0m[mov,mp4,m4a,3gp,3g2,mj2 @ 0x702d28000d00] st: 0 edit list: 1 Missing key frame while searching for timestamp: 1001
01:58:10 [0m[33m[WARN] [0m[mov,mp4,m4a,3gp,3g2,mj2 @ 0x702d3c000d00] st: 0 edit list: 1 Missing key frame while searching for timestamp: 1001
01:58:10 [0m[33m[WARN] [0m[mov,mp4,m4a,3gp,3g2,mj2 @ 0x702d24000d00] st: 0 edit list: 1 Missing key frame while searching for timestamp: 1001
01:58:10 [0m[33m[WARN] [0m[mov,mp4,m4a,3gp,3g2,mj2 @ 0x702d28000d00] st: 0 edit list 1 Cannot find an index entry before timestamp: 1001.
01:58:10 [0m[33m[WARN] [0m[mov,mp4,m4a,3gp,3g2,mj2 @ 0x702d48000d00] st: 0 edit list: 1 Missing key frame while searching for timestamp: 1001
01:58:10 [0m[33m[WARN] [0m[mov,mp4,m4a,3gp,3g2,mj2 @ 0x702d48000d00] st: 0 edit list 1 Cannot find an index entry before timestamp: 1001.
01:58:10 [0m[33m[WARN] [0m[mov,mp4,m4a,3gp,3g2,mj2 @ 0x702d50000d00] st: 0 edit list: 1 Missing key frame while searching for timestamp: 1001
01:58:10 [0m[33m[WARN] [0m[mov,mp4,m4a,3gp,3g2,mj2 @ 0x702d50000d00] st: 0 edit list 1 Cannot find an index entry before timestamp: 1001.
01:58:10 [0m[33m[WARN] [0m[mov,mp4,m4a,3gp,3g2,mj2 @ 0x702d30000d00] st: 0 edit list: 1 Missing key frame while searching for timestamp: 1001
01:58:10 [0m[33m[WARN] [0m[mov,mp4,m4a,3gp,3g2,mj2 @ 0x702d30000d00] st: 0 edit list 1 Cannot find an index entry before timestamp: 1001.

[mov,mp4,m4a,3gp,3g2,mj2 @ 0x702d48000d00] st: 0 edit list: 1 Missing key frame while searching for timestamp: 1001
[... 427 lines cut ...]
01:58:28 [0m[34m[INFO] [0m[b30f19fb] Processing done
01:58:28 [0m[34m[INFO] [0m[57242d85] Processing done
01:58:28 [0m[34m[INFO] [0m[5c4248ee] Processing done
01:58:28 [0m[34m[INFO] [0m[0f97bf7c] Processing done

01:58:28 [0m[34m[INFO] [0m[f1208c9b] Processing done
01:58:28 [0m[34m[INFO] [0m[9fa03517] Processing done
01:58:28 [0m[34m[INFO] [0m[566e48d3] Processing done

[mov,mp4,m4a,3gp,3g2,mj2 @ 0x702d50000d00] st: 0 edit list: 1 Missing key frame while searching for timestamp: 1001
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x702d50000d00] st: 0 edit list 1 Cannot find an index entry before timestamp: 1001.

01:58:28 [0m[33m[WARN] [0m[mov,mp4,m4a,3gp,3g2,mj2 @ 0x702d50000d00] st: 0 edit list: 1 Missing key frame while searching for timestamp: 1001
01:58:28 [0m[33m[WARN] [0m[mov,mp4,m4a,3gp,3g2,mj2 @ 0x702d50000d00] st: 0 edit list 1 Cannot find an index entry before timestamp: 1001.
01:58:28 [0m[33m[WARN] [0m[mov,mp4,m4a,3gp,3g2,mj2 @ 0x702d50000d00] infe version < 2
01:58:28 [0m[33m[WARN] [0mis not implemented. Update your FFmpeg version to the newest one from Git. If the problem still occurs, it means that your file has a feature which has not been implemented.

[mov,mp4,m4a,3gp,3g2,mj2 @ 0x702d50000d00] infe version < 2 is not implemented. Update your FFmpeg version to the newest one from Git. If the problem still occurs, it means that your file has a feature which has not been implemented.

01:58:28 [0m[34m[INFO] [0mOpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_copy_opts cl_khr_gl_event cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_nv_kernel_attribute cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_opaque_fd cl_khr_external_memory_opaque_fd" } Device: NVIDIA Corporation Quadro P400

01:58:29 [0m[34m[INFO] [0mInitialized OpenCL for (1920, 1080, 1920) -> (1080, 1920, 1088), key: BufferDescription { size: (1920, 1080, 1920), rect: None, rotation: None, data: Cpu { buffer: [29, 32, 36, 38, 43, 51, 65, 67, 56, 46, 50, 57, 55, 58, 61, 62, 52, 52, 52, 51, 50, 54, 58, 56, 52, 52, 53, 52, 60, 66, 62, 52, 44, 42, 41, 44, 57, 63, 70, 65, 60, 63, 65, 64, 60, 46, 48, 51, 49, 52, 60, 65, 64, 62, 61, 56, 52, 55, 64, 59, 52, 52, 54, 61, 56, 61, 65, 61, 63, 63, 60, 60, 65, 60, 50, 50, 52, 50, 47, 47, 55, 62, 61, 59, 49, 48, 56, 55, 54, 60, 74, 66, 40, 43, 51, 53, 54, 56, 54, 50, 43, 35, 30, 32, 48, 54, 58, 57, 59, 51, 49, 51, 61, 64, 63, 58, 51, 43, 41, 53, 64, 62, 56, 55, 57, 57, 57, 56, 58, 67, 65, 51, 50, 56, 56, 61, 54, 50, 53, 53, 58, 65, 59, 47, 47, 50, 52, 53, 51, 49, 49, 51, 52, 56, 57, 58, 63, 69, 70, 71, 74, 71, 65, 60, 62, 65, 67, 68, 61, 57, 56, 61, 74, 80, 70, 65, 64, 62, 56, 55, 56, 51, 55, 62, 63, 59, 59, 59, 64, 66, 63, 58, 52, 47, 45, 43, 49, 51, 56, 61, 62, 53, 48, 49, 56, 66, 66, 63, 68, 78, 67, 49, 42, 47, 51, 5 [... cut ...] , 225, 225, 226, 225, 230, 227, 225, 226, 229, 225, 214, 205, 206, 219, 229, 235, 234, 229, 230, 184, 82, 16, 25, 21, 26, 27, 23, 24, 24, 26, 27, 29, 30, 30, 30, 29, 28, 28, 28, 28, 31, 32, 31, 29, 30, 33, 33, 33, 32, 35, 35, 35, 38, 41, 43, 46, 47, 49, 50, 45, 46, 48, 49, 47, 47, 43, 39, 40, 40, 40, 40, 44, 42, 42, 43, 45, 45, 46, 46, 44, 43, 39, 39, 43, 45, 45, 43, 42, 44, 44, 43, 37, 34, 35, 35, 33, 33, 34, 34, 31, 29, 29, 29, 29, 27, 28, 28, 

[Basically a wall of numbers here on out...]
```

jameslin123, 2024-10-11:
In spite of the noise, the CLI process does make progress in encoding the output videos, albeit slowly due to having to log so much text.

AdrianEddy, 2024-10-11:
what was the command line you executed?

jameslin123, 2024-10-11:
This command would repro the issue for me:

`~/gyroflow-dev/Gyroflow-linux64.AppImage /home/james/Videos/A7C2/20241009/20241009_C0400.MP4 --preset "{'output':{'bitrate':23,'codec':'H.265/HEVC',},'stabilization':{'smoothing_params':[{'name':'smoothness','value':0.25},{'name':'smoothness_pitch','value':0.5},{'name':'smoothness_yaw','value':0.5},{'name':'smoothness_roll','value':0.5},{'name':'per_axis','value':0},{'name':'trim_range_only','value':0},{'name':'max_smoothness','value':1},{'name':'alpha_0_1s','value':0.1}],},'version':2}" --overwrite`

On my Ubuntu machine, turns out the bug happens for both CLI and GUI; if I launch GUI and start the rendering there, the terminal still outputs the same huge amount of noise. And the GUI is reporting 5 days of ETA instead of the previous couple hours, due to the time spent on logging text.

On my Windows 10 machine, the GUI doesn't output any logs since I don't launch it from a terminal, but because the ETA is relatively normal, I assume the problem doesn't exist on Windows.

AdrianEddy, 2024-10-11:
I reduced the noise in latest dev build (gh2174) - https://github.com/gyroflow/gyroflow/actions/runs/11285594104/artifacts/2043529990

But your preset is invalid, it should be 
```js
{'output':{'bitrate':23,'codec':'H.265/HEVC'},'stabilization':{'smoothing_params':[{'name':'smoothness','value':0.25}]},'version':2}
```

closed 2024-10-11 by AdrianEddy