gyroflow/gyroflow #906: Lens Support for Sony A7CII kit 28-70mm

Bristopher, 2024-10-14:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Not sure if this is the correct place for a lens support request, sorry if it's not feel free to delete. I was wondering if anyone could add lens support for Sony A7CII kit 28-70mm 4k 16by9 3840x2160/1920x1080-23.98fps. I tried using "Sony_A7III_kit FE 3.55.6 28-70 55__1080p_16by9_1920x1080-100.00fps" and modifying the resolution and fps because it's so similar but it resulted in a lot of stepping in the stabilization. Looking forward to LensFun database support because it supports this lens, thanks!

AdrianEddy, 2024-10-14:
Use the dev build and it's automatic https://gyroflow.xyz/devbuild/

closed 2024-10-14 by AdrianEddy

Bristopher, 2024-10-14:
Fantastic thanks!