gyroflow/gyroflow #908: -g, --gyro-file Option Not Functioning; Synchronization Operation Not Performed

steinvenic, 2024-10-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

Gyroflow 1.5.4 (gh2191)

### What operating system are you using?

20.04.5 LTS & Win10 &Win 11

### What happened?

Everything seems normal only when the preset is saved as the default configuration.
[image]
[image]

AdrianEddy, 2024-10-21:
but `-g` is for the gyro data file, not for preset. Remove `-g` from the command line, just specify the preset without any prefix

steinvenic, 2024-10-22:
Thank you for your patient response. I now understand how to use it.
`./gyroflow /opt/Gyroflow/data/20241017102421_000009.MP4  --preset /opt/Gyroflow/data/gyroflow -f`

closed 2024-10-22 by steinvenic