gyroflow/gyroflow #909: Colors export are different while using GPU encoding or not

tipaul426, 2024-10-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.0

### What operating system are you using?

Windows version 22H 19045.5011

### What GPU are you using?

RX 5700 XT

### What happened?

Hi,
After the 1.6.0 update rendering is different if I use Enconding CPU or not.
If I export a footage with CPU encoding : 

Before : 
[image]

After :
[image]

There will be no problem if without performance encoding it is a fast export but it's taking ages if you dont tick the parameter.
Do I need to untick something so even with CPU Encoding there is no colors changed between before and after ?

Already tried to reset to default settings but it's not better.

Summary : if I export without CPU encoding I have the original colors, if I do export with, colors are modified.
I tried to also reinstall my GPU but not better.
Didn't have that issue with the previous version.
Already reinstalled twice Gyroflow.

Im available if needed;

Paul

PS : camera parameter
[image]

AdrianEddy, 2024-10-25:
this looks like some LUT applied, are you sure you're checking correct files? It would be very weird because Gyroflow doesn't apply any color changes, and the only issue that could be related to this (color range) won't look like that for sure

AdrianEddy, 2024-10-25:
I just tested with Avata 2 (D-LogM) and RTX 3080 Ti and no problem here. Looks like same parameters as yours
[image]

tipaul426, 2024-10-25:
Hey, thanks for the answer, let me show it to you by video :
https://github.com/user-attachments/assets/716abcf2-9bd2-4eb1-b682-d06cff9649c6

Looks really weird, is there a way to rollback the version maybe to 15.0 to see if I can reproduce that ? 
I exported like 50+ video from my avata without having the issue

ty

AdrianEddy, 2024-10-25:
ok, can you send me gyroflow.log file? it's in `%localappdata%/Gyroflow` (paste that in file explorer dir path)

tipaul426, 2024-10-25:
here we go :
[gyroflow.log](https://github.com/user-attachments/files/17523147/gyroflow.log)

AdrianEddy, 2024-10-25:
ok, can you please check this version and also send me gyroflow.log after rendering on this one?

https://github.com/gyroflow/gyroflow/actions/runs/11254703273/artifacts/2034788248

tipaul426, 2024-10-25:
Same behavior :
[gyroflow.log](https://github.com/user-attachments/files/17523397/gyroflow.log)

i dont recall on which version I was before updating today ..
May it be a weird parameter ticked ?

tipaul426, 2024-10-25:
just to let you know I tested some old footage that I had w/ same parameters on my avata with D-LOG M also, and yet the same issue is hapenning, so I dont think it's coming from the avata

AdrianEddy, 2024-10-25:
interesting, that version was before a change which I thought could have influenced this. But since it's not, we need to find first version which broke. 

Please go to https://github.com/gyroflow/gyroflow/actions and start trying older and older versions until you find one that works. After clicking on a version, scroll down to Artifacts and then Gyroflow-windows. You can start from eg. # 2147, and then go lower like # 2122 and so on. The one I sent you in previous message was 2158

tipaul426, 2024-10-25:
Hey,
Just tested out a bunch of version,
Looks like the issue is happening between version 2100 and 2101,
Export is ok on 2100 but not OK on 2101,

Let me know if you need more details, I will gladly help you ;)

AdrianEddy, 2024-10-25:
ok so something changed in ffmpeg

can you try copying these files from 2100 to the latest one (2206)?
 ```
 avcodec-61.dll
 avdevice-61.dll
 avfilter-10.dll
 avformat-61.dll
 avutil-59.dll
 swresample-5.dll
 swscale-8.dll
 ```

tipaul426, 2024-10-25:
I can confirm that version 2206 is working fine when you replace those dll thanks !

AdrianEddy, 2024-10-25:
can you also send me a log from the latest version with these new files where it's working?

AdrianEddy, 2024-10-25:
I suspect it's one of these commits to ffmpeg:
[image]

it's specific to AMD encoder

is the Avata video HDR?

AdrianEddy, 2024-10-25:
Also, can you send me your original video file?

tipaul426, 2024-10-25:
here are the logs :
[gyroflow.log](https://github.com/user-attachments/files/17524382/gyroflow.log)

Im filming in 4K/60fps with DLOG-M on, not sure if this is enabling HDR by default

tipaul426, 2024-10-25:
> Also, can you send me your original video file?

Yep I can, but its 1GB size, where do you want me to upload it ? thanks

AdrianEddy, 2024-10-25:
you can use wetransfer, google drive, dropbox, mega or swisstransfer

tipaul426, 2024-10-25:
https://drive.google.com/file/d/1OFyV08QcJnrK8fNYCeQt2jTvW-Olt5Dw/view?usp=sharing

AdrianEddy, 2024-10-25:
ok looks like I can reproduce, but the issue is in the metadata and depends actually on the player
In MPC-HC I see the original colors, but in Windows' Media Player, I see broken colors. Please try different players (eg. VLC, mpv, MPC-HC, some video editor like Resolve or Premiere if you have) and confirm

tipaul426, 2024-10-25:
Okay, I used VLC > Broken colors
If I upload the video with the broken colors to an editor like Filmora or Capcut I have the same behavior

tipaul426, 2024-10-25:
Tried Microsoft Clipchamp to see, I have the right colors so it really depends on the player

AdrianEddy, 2024-10-25:
Ok this is clearly a bug in ffmpeg, I reported it [here](https://trac.ffmpeg.org/ticket/11269#ticket). I hope you don't mind if I used your sample video

tipaul426, 2024-10-25:
Sure no problem !
Have a nice day

tipaul426, 2024-10-25:
Should be fine now

AdrianEddy, 2024-10-28:
could you try latest dev build? https://gyroflow.xyz/devbuild/

tipaul426, 2024-10-28:
hey, can confirm that it's working fine now !
Thanks

closed 2024-10-28 by AdrianEddy