gyroflow/gyroflow #910: Corrupted A7IV footage

hime87, 2024-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.0

### What operating system are you using?

MacOS 15.0.1 

### What GPU are you using?

M1 Pro

### What happened?

Gyroflow corrupts A7IV footage after stabilisation. Each exported video has multiple frames missing (or frozen) at the beginning. Everything was ok on 1.5.4. With footage from DJI Avata 2 it's not reproducible. I also tried nightly build - same issue. In addition it seems that playback starts little bit choppy each time - so I assume, that export settings are irrelevant.

[image]
[image]
[image]

AdrianEddy, 2024-10-27:
This seems related to 4:2:2 10-bit **de**coder on certain macOS devices. You should be able to workaround this by disabling GPU **de**coding, which is in Advanced on bottom right
[image]

hime87, 2024-10-27:
> This seems related to 4:2:2 10-bit **de**coder on certain macOS devices. You should be able to workaround this by disabling GPU **de**coding, which is in Advanced on bottom right [...]

Thanks, I just unchecked it and I can confirm that it's not reproducible :)

AdrianEddy, 2024-10-27:
also, since you're using Premiere, you can just use the Gyroflow plugin for premiere instead of rendering. There's a `Video editor plugins` panel in the app on bottom right for easy installation

AdrianEddy, 2024-10-31:
I can't reproduce this on M2 Air. Can you send me gyroflow.log file? it's in `$HOME/Library/Application Support/Gyroflow/`
Render with GPU decoding (so it's corrupted), then close gyroflow and then send me gyroflow.log file

Maybe you can upload your sample video file too?

wewps, 2024-11-01:
First off, I just want to say the new support for IBIS enabled shots is amazing. It such a relief to not have to worry about whether or not it was toggled off for a clip that needs stabilization in post!

I also started experiencing dropped/frozen frames and artifacts after export just like OP, but with 8-bit 4:2:0 footage. Everything looked good previewing in app- the issues only appeared on the exported version. Disabling GPU decoding also seems to have completely fixed the errors here as well despite mine being much tamer 8 bit 4:2:0 rec709 footage. 

For reference: M1 Air, MacOS 15.2 beta

hime87, 2024-11-01:
> I can't reproduce this on M2 Air. Can you send me gyroflow.log file? it's in `$HOME/Library/Application Support/Gyroflow/` Render with GPU decoding (so it's corrupted), then close gyroflow and then send me gyroflow.log file
> 
> Maybe you can upload your sample video file too?

Sure Adrian, thanks for Your support and amazing software!

Link to original, stabilised footage and log:
[https://we.tl/t-rzAQbqE2zG](url)

AdrianEddy, 2024-11-01:
Since this is an error from a hardware decoder, there's nothing I can do about it, other than automatically try decoding without GPU in case of this error. Which is exactly what I did in the latest dev build now https://gyroflow.xyz/devbuild/

closed 2024-11-01 by AdrianEddy