gyroflow/gyroflow #913: Rotation attribute is ignored during merging

apog78, 2024-11-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.0

### What operating system are you using?

Ubuntu 24.04

### What GPU are you using?

AMD Ryzen 5 5600G

### What happened?

When merging multiple files, the rotation attribute is ignored.  Imagine some video were recorded overhead (rotation 180) and upon merging, since no error was detected, GyroFlow offers to remove the original files..  Not good.
Since rotation attribute is regarded when a video is opened separately, I consider this a bug.
Thank you very much for the great product!
With best regards, apog78

AdrianEddy, 2024-11-01:
Can you send me a sample unmerged files to reproduce?

apog78, 2024-11-01:
Sure I can, but 2 files I have issue with are 500MB in size.  I think we can do better -- if you have any video files yourself, let's just rotate them using `ffmpeg -i in.mp4 -vf "transpose=2,transpose=2" out.mp4` (transposing two times to get 180 deg).  This must be done to both input files.  Now, try to merge them.  Thank you!

apog78, 2024-11-01:
I guess, you also have to "fix" the orientation tag.  Do it like this: `exiftool -Rotation=3 file.mp4`

AdrianEddy, 2024-11-01:
well, I can't reproduce it like that, it works as expected for me. The rotation tag is retained
And I really don't see why it wouldn't work, as all the original metadata and tags are preserved when merging. 

Send me your files where this is reproducible

apog78, 2024-11-02:
I narrowed the issue to this -- merging two files with the first one "normal" and the second one "bottom up". If you still can't reproduce, I will send you files.  Thanks!

AdrianEddy, 2024-11-02:
hmm and what do you expect to happen in this case? Have the first video like original and the second part rotated? That's not possible because we're not re-encoding the videos, so it can only have one rotation flag for the entire merged video

apog78, 2024-11-02:
Well, of course a naive user (like me) expects that the resulting video is consistent with rotation=0 (upright).  If this is technically not possible (why would a normal user even know what "re-encoding" means), he/she must be at least warned, that the orientation flag of SOME parts cannot be taken in account and the resulting video is going to be wrong and useless.  Why spend time joining parts if the result is unusable?  Even worse -- offer to remove the originals knowing that the result is probably broken and cannot be easily fixed (think about it).  Thank you!

AdrianEddy, 2024-11-02:
You're right, I should at least check resolution, framerate and rotation and prevent users to merge the files if these don't match. I updated that in latest dev build https://gyroflow.xyz/devbuild/

closed 2024-11-02 by AdrianEddy

apog78, 2024-11-02:
OK, great, but why not ask a user to rotate the vids which are marked as "bottom up"?

AdrianEddy, 2024-11-02:
The file merger in Gyroflow is designed only for one purpose - to join a single video that has been split in camera because of filesystem size limits. It's not designed to merge arbitrary videos, that's way out of scope and way more complex to implement.

The merger is "dumb", it doesn't understand the videos at all, it just merges the raw data and that's it. If you want to merge arbitrary videos then you have to use ffmpeg or a proper video tool which can reencode the videos.

Asking the user to just rotate the videos is wrong, because the user shouldn't do such things in the first place, hence now it throws an error instead of letting you proceed with the merging.

apog78, 2024-11-02:
Thank you, this is a clear answer!  Why the merger must stay "dumb" is a mystery, though.

AdrianEddy, 2024-11-02:
It's very simple: 
- We can't make your parser more smart because that's an insane amount of work to handle all formats, codecs and edge cases. 

- We can't use ffmpeg for this purpose because it discards gyro data and other needed metadata. We can't make it preserve all the metadata, because it's not designed to do it and there are too many edge cases which break in weird ways.

Gyroflow's merger is written for one purpose and it does it's job perfectly