gyroflow/gyroflow #914: Warping on my SONY A7CⅡ with Tamron 150-500mm F/5-6.7 Di III VC VXD

booknary, 2024-11-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.0 & 1.6.0(gh2226)

### What operating system are you using?

Windows 11 24H2

### What GPU are you using?

NVIDlA GeForce GTX 1650(laptop 4G)

### What happened?

Hi, Eddy!
I noticed that you have posted one issue [here](https://github.com/gyroflow/gyroflow/issues/849#issue-2375878122), I am curious if the second one "_Warping in some cases, possibly related to IBIS/OIS/EIS but it could also be the lens profile mismatch with EIS mesh (C1877, C1363)_" is the same as mine. Here are the videos:
Original: https://drive.google.com/file/d/1_AkDl-EFhPUeXNc3n0cK4RVRCCM5o_CB/view?usp=sharing
Processed: https://drive.google.com/file/d/1IvkWeqgoeFF7VCh9tGKsXBE-xDb37CQq/view?usp=sharing
There is some image distortion in specific areas after stabilization.
Here is the warning and setting:
[image]
[image]
Thank you for your help!

booknary, 2024-11-03:
And several of the output work well, I'm not sure why this happened.
Works here:
Original: https://drive.google.com/file/d/1Co8ohYHvVekO3XbnKaSNOdvUESVPh60_/view?usp=sharing
Processed: https://drive.google.com/file/d/1OhfTzBy-KDcaQAYEs3o6qM0htF9t86Eu/view?usp=sharing

closed 2024-11-03 by AdrianEddy

AdrianEddy, 2024-11-03:
Your file is particularly interesting because even Catalyst creates this warping. It looks like a bug in the camera, because I can avoid the warping by ignoring the mesh distortion data, but that's not a right thing to do, so something seems wrong in the metadata here

booknary, 2024-11-04:
I got it. I was going to try Catalyst, but I found some post say it doesn't work for non-Sony lens. Thank you very much! And I am glad my files can help.