gyroflow/gyroflow #915: Gyroflow background mode not working correctly. It is not fill right and top part of the video. Can you advise me how to fix it? (I tested also a nightly build release too).

AndriiShramko, 2024-11-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.0

### What operating system are you using?

Windows 10 Pro 22h2

### What GPU are you using?

rtx 4090

### What happened?

Gyroflow background mode not working correctly. It is not fill right and top part of the video. Can you advise me how to fix it? (I tested also a nightly build release too). 
Thanks. https://youtube.com/shorts/adYbgwn5wG0?si=PETO4pfsUpbgE4uS

AdrianEddy, 2024-11-04:
can you send me the video you're using? does it also happen in the app, or only in the plugin?

AndriiShramko, 2024-11-04:
only in plugin

AndriiShramko, 2024-11-04:
I did a test with video from different cameras and have the same problem.

AndriiShramko, 2024-11-04:
I can give you a teamviewer and you can check it on my PC. zmei116@gmail.com

AdrianEddy, 2024-11-05:
Fixed https://gyroflow.xyz/devbuild/

closed 2024-11-05 by AdrianEddy

AndriiShramko, 2024-11-05:
Amazing! Thank you Adrian. You are the best!