gyroflow/gyroflow #919: Open file in Video information vs Motion data

VBK7, 2024-11-06:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

My workflow:
I record ProRes Raw on a Ninja V connected to an A7S3 which is simultaneously recording standard XAVC S video.
I then use the gyro data from the XAVC S files to stabilize the ProRes Raw footage.

The problem:
When I open XAVC S files in the [Video information] section of the program, Gyroflow automates everything including gyro data offset correction, lens profiles, and rolling shutter correction. Good 👍

When I open a proxy ProRes Raw file in the [Video information] section, and then open the corresponding XAVC S file in the [Motion data] section of the program, Gyroflow throws all that information (gyro data offset correction, lens profiles, and rolling shutter correction) out the window, and only uses the gyro data from the file. Bad 👎
Because of that,
1. I have to manually correct the gyro data offset. There's no documentation or way to tell how much gyro data offset correction Gyroflow applies automatically to these supported Sony files so I have to guess.
2. I have to create my own lens profiles. There's no way to apply the automatic lens correction profiles from the XAVC S file.
3. I have to manually put in the rolling shutter correction value.

The reason I have to work this way is because the ProRes Raw footage is a different resolution than the XAVC S footage (4264x2408 vs 3840x2160). If I create a Gyroflow project where I open a XAVC S file directly in the [Video information] section and then load that Gyroflow project file in the DaVinci Resolve OpenFX plug-in onto a ProRes Raw clip, it doesn't work because of the resolution mismatch.

2 ways to fix this;

The easy way:
Make the [Open file] button in the [Motion data] section work the same way as the [Open file] button in the [Video information] section of the program. In other words, so it would automatically apply all the corrections that are available in the file being opened.

The more elegant way, here's an example:
✅ Gyro data offset correction
🔳 Use lens profile
✅ Rolling shutter correction
When a user will open a file, Gyroflow can then display which automatic corrections are available as a list of checkbox options. The user can then check or uncheck which automatic corrections they would like Gyroflow to apply. You could create a new section on the left side of the program called something like [Automatic corrections] where these checkbox options would appear whether you open a file in the [Video information] section or the [Motion data] section of the program. That way, you wouldn't have to have the same options appear in both the [Video information] section and the [Motion data] section of the program.

The gyro data offset correction and the rolling shutter correction would work perfectly because the ProRes Raw footage retains the same rolling shutter and is frame for frame the same as the XAVC S footage timewise. The only minor problem would be applying the lens profile because of the resolution difference. The ProRes raw records a tiny bit more of the sensor vertically. Would be great if I could still apply the lens correction from the Sony metadata to the proxy ProRes Raw file.

Thank You!

AdrianEddy, 2024-11-06:
are you sure that the ProRes file is exactly frame-to-frame accurate to the XAVC-S file and that is consistent across many recordings? I find that hard to believe, because these files are recorded by different devices and Ninja only sends the recording trigger to Sony without any guarantee how long it will take to start recording. For stabilization even 1 frame is a big difference

Rolling shutter correction could be loaded automatically, but you can already use the default preset for that. Also, you have to remember that this is very specific to your use case and every time I implement something specific to someone's case, I have a lot more reports from people that it's not a great feature because their use case is different. That happened a lot of times already

There's nothing I can do for the lens profile since the resolutions are different. You could use the one from XAVC S and accept the inaccuracies, but I can't make such behavior default or even recommended

Using the plugin and simply applying it on the ProRes file and using "Browse" to select XAVC file is the best option for your use case in my opinion

AdrianEddy, 2024-11-06:
A common use case is using eg. DJI Action as a gyro logger for eg. Canon camera, and in that case loading offset, lens profile or even rolling shutter correction is a wrong thing to do. 

I could make an exception that if Sony camera is loaded as gyro source then it's probably your use case, because I can't think of using a Sony camera to record gyro data for another cinema camera which has different parameters. Oh right, there's Sony RX0

Anyway, while I'm not a fan of custom whitelists like that, I could probably make it like that with the exception of Sony RX0

VBK7, 2024-11-06:
Yes, 100%. The clips always match perfectly. I can’t remember exactly right now, but I remember there is some kind of trigger recording option in the Ninja V that allows it to communicate with the camera (not sure exactly what it does) and the recordings can be triggered from either the Ninja V or the camera. As far as I understand, the XAVC S files are just downsampled files of the raw output with corrections applied.

The rolling shutter is the least of my issues as I know what the default correction value is when I open the file in the [Video information] section, but it would be nice if that could be applied automatically. The bigger problem is that I don’t know how much automatic gyro data offset correction Gyroflow applies to supported Sony files so I can’t just copy that number. If you implement the 2nd fix, I don’t see how that could be a problem for other people as you could have all options off by default. This is also not the only workflow I’ve had this problem with. The lens profile for the DJI Osmo action 4 is automatically applied, and it’s not possible to search it and apply it from the database. I’ve had instances where I applied noise reduction in DaVinci and then opened that file in the [Video information] section and the original file in the [Motion data] section and there was no way to apply the lens profile because it is only automatically applied when you open it in the [Video information] section of the program. There were many other times where the fact that being forced to open the file in one or another section of the program means that you get completely different results was a big issue. I see it as a fundamental problem. There’s no need to force anything on users that are not affected by this problem, just give us the on/off option when we need it.

I’m completely fine with using the lens profile from the XAVC S file with inaccuracies. Please make it possible though. How does Gyroflow handle applying lens profiles with different resolutions? What it be possible to add options where you can choose to apply the lens profile in a fit, fill, or crop manner when resolution doesn’t match? Like in DaVinci Resolve or FCPX when you’re making a file with different resolution fit into the project/timeline resolution. That would fix most resolution mismatch problems.

VBK7, 2024-11-06:
> A common use case is using eg. DJI Action as a gyro logger for eg. Canon camera, and in that case loading offset, lens profile or even rolling shutter correction is a wrong thing to do.
>
> I could [...]

This defeats the convenience of recording the Gyro data in Camera.

VBK7, 2024-11-06:
> I’m completely fine with using the lens profile from the XAVC S file with inaccuracies. Please make it possible though. How does Gyroflow handle applying lens profiles with different resolutions? Would it be possible to add options where you can choose to apply the lens profile in a fit, fill, or crop manner when resolution doesn’t match? Like in DaVinci Resolve or FCPX when you’re making a file with different resolution fit into the project/timeline resolution. That would fix most resolution mismatch problems.

Any thoughts on this?

AdrianEddy, 2024-11-06:
Not sure about this at the moment, but it would help if you can send me a pair of ProRes/XAVCS videos with calibration pattern

AdrianEddy, 2025-01-20:
I added "Load all metadata" checkbox in Motion data section in ffb9556bc9986e413b049aa076edfc35e240f702

[image]

Tested on your file, looks good to me