gyroflow/gyroflow #920: Zoom smoothing

VBK7, 2024-11-06:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

With Gyroflow now perfectly working with zoom lenses and Sony files, would be super neat to add a smoothing algorithm to abrupt zooms _|   / to make them look more like a premium camcorder zoom. Since Gyroflow knows the focal length and lens distortion at each focal length, this seems like an obvious feature to add. Basically a separate stabilization specifically for the change of focal length (zooming) in a video. Any plans on implementing this in the future?

AdrianEddy, 2024-11-06:
yes in theory it's possible, but not sure how much work it is to implement. I'll think about it

VBK7, 2024-11-08:
I've been thinking about this the past few days and there's a few more things to consider like the fact that the difference between 10-20 mm is much bigger visually than the difference between 90-100 mm. 10-20 mm = 2x magnification and 90-100 = 1.1111x magnification, even though they are both a 10 mm difference. Rolling shutter on fast zooms could also be fixed. Theoretically, the bottom of a frame would be more magnified than the top if you zoom really fast.

cerrec, 2025-03-30:
FWIW, the Similarity stabilizer option in Davinci does just this - it smooths out a manual zoom, and does a very nice job. So, it's definitely possible.

Milchgsicht-star, 2025-10-25:
This would be such a cool feature!

VBK7, 2025-10-25:
Would also be nice to have focus breathing compensation. From what I’ve seen, if you turn it on in Sony cameras, it seems they apply a magnification value for the point at which the focus compensation would have to be corrected the strongest and reduce the magnification for the rest of the focus range throughout the video appropriately. The problem is that you don’t always hit that focus range at which the most compensation has to be applied so the camera is unnecessarily digitally magnifying the entire clip too much for no reason. The correct way to do it would be to look at what point in the video clip the focus has to be corrected the most, and base the rest of the correction for that clip off of that point. That way, the digital magnification amount for each clip would be different because you would only be correcting for as much as you need to correct for. Say you have a locked off interview shot on a tripod, the focus is barely going to be changing within a very small zone, so there would need to be very little digital magnification applied for the focus breathing compensation. Whereas if you’re taking a handheld video and focus on something very close and then back to near Infiniti focus in the same video clip, there will have to be much more digital magnification applied throughout the whole clip for the focus breathing compensation. It seems that by default, Sony cameras aren’t taking advantage of this and are by default applying the highest digital magnification value for the current lens and then reducing it as necessary throughout the clip. Would be nice to have this feature in Gyroflow, that way we would have much more control over it and could also apply it to ProRes RAW footage from Sony cameras.

AdrianEddy, 2026-04-23:
Can you share a sample file?