gyroflow/gyroflow #921: Drifting issue when stabilizing Sony A7 IV

tivikoff, 2024-11-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.0

### What operating system are you using?

Win10 x64

### What happened?

Hi,
I noticed strange drifting issue when stabilizing my Sony A7 IV footage. 
You can see that Gyroflow stabilized output has drifting pan motion which was not present in the source footage:

https://github.com/user-attachments/assets/8b5d368b-9212-45b2-800b-1acf69477b82

Here is Gyroflow output overlayed on the Source footage with difference mode showing large drift from top-left to the center and then to the right:

https://github.com/user-attachments/assets/582201b1-09d5-4ed0-83b9-dcf892e016a6

And here is just to compare how it looks with the Catalyst output without any drifting - stays in the center:

https://github.com/user-attachments/assets/f0f86f6c-9f28-4ad4-8e0d-102783d104d3

While this drifting issue is much more noticeable with less shaky footage, I made another one with the large shaking. 
As you can see, Catalyst stabilized all large shakes while keeping original position with adding extra motion:

https://github.com/user-attachments/assets/7fe2ac35-0546-4a07-b3c8-ad3e55d8e62b

Gyroflow also stabilized large shakes pretty well, but again - it adds unneccessary pan right from the beginning:

https://github.com/user-attachments/assets/2f6f6512-5c95-4f0e-8be7-cd5e8c57d42a

As you can see - again from the top-left to the center, and then to the right. And this vector of movement each time has the same "pattern" for each footage.

Tested with different settings, smoothness etc. (Higher smoothness - larger drift), static zoom, or without zoom at all. All the same. Each time this drift, which ruins static shots by adding this strange panning.
Tested without IBIS/OIS/EIS - all disabled in camera. Latest camera firmware (4.0). Tamron 28-75mm G2 (tested with full range of focal distances - all the same). Gyroflow with default settings (tried to full reset as well).
This issue is not new, I noticed this since 1.5.0 or even 1.4.x.

GoPro Hero 11 works perfectly. But not Sony A7 IV.

If that's not a bug and Sony needs some special settings which I missed - I'd be very grateful if you tell how to fix this. 

You can find source footage as well as my Catalyst and Gyroflow stabilized outputs here:
[Small Shake](https://drive.google.com/drive/folders/1QYxaDxgB1DxbIriTE8vPBVfon2ppJhZ2?usp=drive_link)
[Large Shake](https://drive.google.com/drive/folders/1CiJi1pH3fHOLQpZtw03WyAEpUsq0O3qW?usp=drive_link)

Thank you very much for your work. Your recent release v1.6.0 with extended Sony metadata support is amazing. 
Can't wait to use it with my Sony camera after we find out how to deal with this drifting issue.

AdrianEddy, 2024-11-10:
Simply try out different Integration methods in Motion data section

closed 2024-11-10 by AdrianEddy

tivikoff, 2024-11-10:
@AdrianEddy Thank you for quick reply. Maybe I'm doing something wrong, but that's the first setting I've tried. And all other methods giving even worse results with larger drift and huge zoom in.

AdrianEddy, 2024-11-10:
right, in your case, it helps to estimate the gyro bias. In general you want to do this at time where the camera is stationary (not moving at all), right click on that moment and "Estimate gyro bias here". If there's no such moment (the camera is moving all the time) then just click anywhere and do that

tivikoff, 2024-11-10:
@AdrianEddy Thank you, this helped! Not completely, but almost eliminated drifting in my case.
Wondering if Sony stores their gyro bias information somewhere in metadata too since their tool somehow compensates it automatically without the need to pre-shoot stationary camera. But this is good workaround anyway. Thanks a lot!