gyroflow/gyroflow #925: Lens doesn't support OSS metadata

simonwjcai, 2024-11-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.0

### What operating system are you using?

MacOS 15.1

### What GPU are you using?

M1

### What happened?

I'm using FX6 with Sony 24-105/F4 zoom lens, with OSS on. However, I've got the following error message, cannot be processed. I've checked the latest update of 1.6.0, saids it can support Zoom len with OSS. Do you have any idea?

Message: "This video cannot be stabilized, because this lens doesn't support OSS metadata. Disable lens stabilization (Optical SteadyShot) in order to use Gyroflow."

[image]

AdrianEddy, 2024-11-13:
It's not supported by the lens, we can't do anything about it

closed 2024-11-13 by AdrianEddy

dizzybro, 2024-11-19:
I noticed that I get this error when the standard stabilization is selected on camera. If I switch to active, I do not get the error. Not sure if this is a limitation of tha hardware or not...

rgaufman, 2025-09-01:
I'm on a Tamron 50-300mm F/4.5-6.3 Di III VC VXD using a Sony ZV-E1 and Sony A7C cameras. I'm also getting this message, but other lenses without OSS work beautifully, even with camera IBIS. There's no way to my knowledge to turn off OSS without also turning off IBIS which adds a lot of motion blur.

Is this something that can be added in the future like Sony IBIS was, or is it simply impossible to add support for? - any known workarounds? Does enabling "active" work properly, or does it introduce its own shake? -- I'm currently using warp stabiliser in Final Cut which has mixed results.

Here is an example clip: https://drive.google.com/file/d/1lrEFin5JQ2hjRAxkNW6qarthE6ywaBhK/view?usp=sharing

AdrianEddy, 2025-09-01:
It's not possible to add it because there's no OSS metadata in this case

I **think** (but I haven't verified that) that the "Active" mode in this case is IBIS + EIS. While it's not possible to disable OSS directly, I think Active mode does that behind the scenes, so that may be a good mode to record in

rgaufman, 2025-09-01:
Thank you I’ll try that, that’s a damn shame though as the OSS makes quite a big difference to reduce motion blur when zoomed i! - it’s one of the main reasons I bought it over the non OSS version. Is this something I can complain to Tamron about? Or is this a Sony limitation?

 I’ll see if IBIS+EIS+Gyroflow works better than IBIS+OSS+WarpStabe.