gyroflow/gyroflow #926: Build iOS release issue

bias123456, 2024-11-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

a76cabda

### What operating system are you using?

MacOS 14.4.1 (23E224), Apple M1

### What happened?

I encountered an issue while compiling the project to iOS release, which seems to be an error of not being able to link to the QT library. 

"ld: symbol(s) not found for architecture arm64 clang: error: linker command failed with exit code 1 (use -v to see invocation)", 

I cannot find the reason or solution. Could you provide some guidance or solution. The [attachment]
[error_log.txt](https://github.com/user-attachments/files/17771528/error_log.txt)
 is the content of the full log on my Mac mini (MacOS 14.4.1 (23E224), Apple M1 ) . Thanks

bbiasxie@BBiasdeMac-mini gyroflow % just ios run
just -f _scripts/ios.just run
   Compiling proc-macro2 v1.0.89
   Compiling unicode-ident v1.0.13
   Compiling libc v0.2.161
   Compiling autocfg v1.4.0
   Compiling shlex v1.3.0
   Compiling cfg-if v1.0.0
   Compiling memchr v2.7.4
   Compiling regex-syntax v0.8.5
   Compiling serde v1.0.214
   Compiling glob v0.3.1
   Compiling num-traits v0.2.19
   Compiling aho-corasick v1.1.3
   Compiling clang-sys v1.8.1
   Compiling quote v1.0.37
   Compiling syn v2.0.87
   Compiling jobserver v0.1.32
   Compiling regex-automata v0.4.8
   Compiling paste v1.0.15
   Compiling cc v1.1.35
   Compiling byteorder v1.5.0
   Compiling libloading v0.8.5
   Compiling stable_deref_trait v1.2.0
   Compiling bytemuck v1.19.0
   Compiling log v0.4.22
   Compiling either v1.13.0
   Compiling num-integer v0.1.46
   Compiling semver v1.0.23
   Compiling itertools v0.13.0
   Compiling smallvec v1.13.2
   Compiling num-bigint v0.4.6
   Compiling getrandom v0.2.15
   Compiling matrixmultiply v0.3.9
   Compiling writeable v0.5.5
   Compiling ring v0.17.8
[... 443 lines cut ...]
            "_png_write_rows", referenced from:
                QPNGImageWriter::writeImage(QImage const&, int, QString const&, int, int) in libQt6Gui.a[arm64][216](qpnghandler.cpp.o)
                QPNGImageWriter::writeImage(QImage const&, int, QString const&, int, int) in libQt6Gui.a[arm64][216](qpnghandler.cpp.o)
            "_qt_main_wrapper", referenced from:
                <initial-undefines>
          ld: symbol(s) not found for architecture arm64
          clang: error: linker command failed with exit code 1 (use -v to see invocation)
          

warning: `gyroflow` (bin "gyroflow") generated 2 warnings
error: could not compile `gyroflow` (bin "gyroflow") due to 1 previous error; 2 warnings emitted
error: Recipe `build-and-package` failed with exit code 101
[....] Waiting for iOS device to be connected
[....] Using d24b15f7168c321105e0bc9bc9b753a58eaa015f (D22AP, iPhone X, iphoneos, arm64, 15.4.1, 19E258) a.k.a. 'iPhone-biasTest'.
------ Install phase ------
[  0%] Found d24b15f7168c321105e0bc9bc9b753a58eaa015f (D22AP, iPhone X, iphoneos, arm64, 15.4.1, 19E258) a.k.a. 'iPhone-biasTest' connected through USB, beginning install
[  0%] Copying /Users/bbiasxie/Documents/TestProject/test_gyroflow/gyroflow/_scripts/..//_deployment/_binaries/ios/Payload/Gyroflow.app/META-INF/ to device
[  0%] Copying /Users/bbiasxie/Documents/TestProject/test_gyroflow/gyroflow/_scripts/..//_deployment/_binaries/ios/Payload/Gyroflow.app/META-INF/com.apple.ZipMetadata.plist to device
[ 52%] CreatingStagingDirectory
[ 57%] ExtractingPackage
[ 60%] InspectingPackage
2024-11-15 10:51:50.294 ios-deploy[28811:618699] [ !! ] Error 0xe8000051: Could not inspect the application package. AMDeviceSecureInstallApplication(0, device, url, options, install_callback, 0)
error: Recipe `run` failed with exit code 253
error: Recipe `ios` failed on line 39 with exit code 253
[error_log.txt](https://github.com/user-attachments/files/17771960/error_log.txt)

AdrianEddy, 2024-11-15:
I'm not sure why you're seeing these errors, running `just ios install-deps` and `just ios deploy` should just work, it works fine in CI

run `cargo clean`, delete the `ext` directory and start again, make sure to call  `just ios install-deps` and `just ios deploy`

bias123456, 2024-11-18:
> I'm not sure why you're seeing these errors, running `just ios install-deps` and `just ios deploy` should just work, it works fine in CI
> 
> run `cargo clean`, delete the `ext` directory and start again, make sure to call `just ios install-deps` and `just ios deploy`

In addition running `just ios install-deps` and `just ios deploy`,  do I still need to install  QT environment independently?

AdrianEddy, 2024-11-18:
No

bias123456, 2024-11-20:
Hi, AdrianEddy,

My MacBook is Apple M1 chip with arm64 architecture,  do I need to do some additional configuration? Thanks

AdrianEddy, 2024-11-21:
it should work on M1, you just need to have XCode and iOS platform installed and everything else should be handled  by the script
BTW, why are you compiling iOS version yourself?

closed 2024-11-29 by AdrianEddy