gyroflow/gyroflow #928: Sony lenses with OIS

p-masek, 2024-11-18:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

This is mostly feature request to add more details (if there are known) about supported lenses into Gyroflow documentation. So far in documentation is mentioned below:

"Currently Gyroflow supports reading most of the recorded metadata, which includes IBIS and OIS data. Therefore, you can record with IBIS and OIS enabled. Most newer or high-end lenses with OIS are supported, but there are some lenses where enabling OIS prevents proper stabilization, so check your setup first before enabling OIS."

I have Sony A7CII. Gyroflow works great if I use lens without OIS (no matter if IBIS is enabled or not), but with Sony 24-105 F4 OIS lens I have to completely switch off whole stabilization (IBIS+OIS)  to achieve smoothest GF stabilization. 
When it is stabilization is enabled on the lens, GF stabilized footage still looks bit shaky (it looks like it is the OIS part of whole stabilization is causing that). Overall best results are achieved when camera stabilization is completely disabled (but that brings obviously higher crop of the image which is unwanted). Indeed it would be great to just disable OIS part which would solve the issue, but that is not possible.

I would expect GF stabilization of this lens (when OIS&IBIS is enabled) will work as it is not too old (mine is brand new, but lens has been release in 2018) and as per Sony this lens&body combination should supports "Body–Lens Coordinated Control" stabilization of OIS&IBIS, firmware is up-to-date on both  ( https://support.d-imaging.sony.co.jp/www/cscs/function/compatibility.php?fnc=1002&area=gb&lang=en&ref=1 ) .

Combined stabilization works so I would expect all metadatas from OIS and IBIS will be saved to the footage.

I have compared Gyroflow and Catalyst stabilization and Catalyst seems to be a bit better, but overall is best to disable stabilization completely.

For reference:
Video with stabilization enabled (both OIS+IBIS):
https://drive.google.com/file/d/10Z3RIuguS9n2577h4XyAt3A_yrkZ1oLj/view?usp=sharing
Video with whole stabilization disabled:
https://drive.google.com/file/d/1GgsAPJAGuJYbd2jgMJSvgSXd3g-U93oN/view?usp=sharing

Is there some list of known lenses which works properly? 
Or is age of lense only key which we can use to determine it?
Is there chance that problem is not caused by the lens and it will work better on some other/newer Sony bodies? (I know it is maybe more question to Sony support).
Or it is better to prefer lenses without OIS? (as IBIS and GF works perfectly fine)

AdrianEddy, 2024-11-18:
Unfortunately I don't have a list of working lenses and I wasn't able to find any at Sony

I have a few samples with working OIS data, and *I think* that was 70-200 GM2, but I'm not 100% sure.

Most kit lenses with OSS don't work from what I've seen so far

The documentation is still the most accurate: `check your setup first before enabling OIS.`. I simply don't have any more details than that

p-masek, 2024-11-18:
Thank you @AdrianEddy 
Any chance you could take a look on first clip I have shared? I have compared GF and Catalyst and Catalyst seems to bit better and shakes (probably) caused by OIS are not so severe.  Maybe it is already partly covered by "Fix remaining Sony issues #849".

AdrianEddy, 2024-11-18:
I don't see anything immediately obvious with that file but it's possible it's already covered by #849. OIS data looks ok in that file, so that lens should be supported

Also, Sony has their own share of bugs in firmware and/or Catalyst, I've seen some files that's supposed to work in Catalyst but they produce bad results.

AdrianEddy, 2024-11-18:
There are multiple layers (at least 5) of processing in case of Sony files, that includes: internal rolling shutter correction, IBIS correction, OIS correction, Mesh correction (which includes calculations for OIS and lens distortions) and lens distortion correction 

Not all of them are implemented fully or 100% properly yet in Gyroflow, that's why we still have #849 open. The thing is that they are not separate layers - they work together and if output of one layer is wrong then it affects all others, so it's hard to pinpoint exactly what's going on here

The funny thing is that Catalyst also doesn't implement it properly in all cases. I have one file which I can stabilize better in Gyroflow than in Catalyst, because their logic is also not 100% correct

Also keep in mind that all of this is mostly going in the blind as none of these algorithms or data points are documented anywhere and we're just guessing based on reverse engineering Catalyst

p-masek, 2024-11-18:
Interesting, thanks for you this info. Hopefully, it will be possible to improve combined OIS&IBIS stabilization one day.
Thank you for your hard work. Gyroflow (and especially plugin for Premiere) is really game-changer and it is one of the main reasons I going to switch from Canon to Sony.

closed 2024-11-18 by p-masek