gyroflow/gyroflow #930: iOS profile error

JunWeiAndroid, 2024-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

a76cabd

### What operating system are you using?

macOS Ventura 13.3

### What happened?

when I run `just ios debug`, some errors have occurred.
[image]
[image]
[image]
[image]

closed 2024-11-21 by AdrianEddy

JunWeiAndroid, 2024-11-22:
@AdrianEddy 
I have placed my own profile file in the designated directory, `DevelopmentCertificateSHA` and `DevelopmentProvision`have been set up, but it's not work. Please help me check if there are any errors in the settings. Thank you.
[image]
[image]

AdrianEddy, 2024-11-22:
what's the error?