gyroflow/gyroflow #932: video saturated after exporting it

SamuuFPV, 2024-12-03:
Hi there! my issue is that when i render 10bit footage shot on gopro hero 12 or DJI action 5 pro, using my GPU, the exported video gets super saturated for some reason, it still keeps 10bit but the video is unusable. My specs are r5 3600 and radeon 6650xt (with AMD pro drivers installed), 32gb ram ddr4. I thought that maybe my GPU does not support 4:2:2 10 bit, but i dont get the error message saying this pixel format is not supported
How it should look: 
[image]

how it looks after exporting:
[image]

AdrianEddy, 2024-12-03:
bug in ffmpeg, already fixed in dev build https://gyroflow.xyz/devbuild/

closed 2024-12-03 by AdrianEddy

SamuuFPV, 2024-12-03:
Now what happens is that i gat the error: this pixel format is not supported (YUV420P LE) i click on p010le and i get this error: 
[image]

SamuuFPV, 2024-12-03:
I always have the boubt that if my GPU does support 10bit or not. i guess it does because it was rendering it before and maintaining 10bit

AdrianEddy, 2024-12-03:
try to reset settings to defaults (in advanced on bottom right) and try again. Also make sure you have latest AMD drivers

SamuuFPV, 2024-12-03:
sure, on it. sorry i thought this was closed and i opened another issue. Do you recommend normal AMD drivers or the Pro edition? i was using the normal ones and thied pro to see if the issue dissapeared

AdrianEddy, 2024-12-03:
I don't know the difference between them, I think both should work

SamuuFPV, 2024-12-03:
Its working Brother! Thank you so much for your help! Ive been dealing with this issue for a long time and always using CPU only.