gyroflow/gyroflow #934: Failed to process pixels: NoCachedWgpuInstance during export

raintonr, 2024-12-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

Gyroflow 1.6.0 (gh2204)

### What operating system are you using?

Fedora Linux 41 (Workstation Edition)

### What GPU are you using?

AMD Ryzen 7 PRO 7840U w/ Radeon 780M Graphics

### What happened?

Using the AppImage. Trying to process footage from DJI Avata 2 (4k/60). I can open files and preview them just fine. GUI preview is very smooth so assume everything is working well with graphics drivers, etc.

However, when I try to export seem to get the first frame (slightly muted colours for some reason) then just ghosting movement.

GUI shows everything was successful with the export when it was not.

Terminal (from where Gyroflow command was launched) shows logging below, seemingly repeated for every frame.

I tried exporting in different codecs, using GPU and not with no change in the result (aside from with no GPU use, the second `DEBUG` log line below is not shown).

### Relevant log output

```shell
21:32:32 [ERROR] Failed to process pixels: NoCachedWgpuInstance("1373832734|opencv_fisheye||8|0|(3840, 2880)|(3840, 2160)|Lanczos4|ThreadId(2)")
21:32:32 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Uploading frame to the device, hw_upload_format VAAPI, final_frame.format: P010LE
```

raintonr, 2024-12-06:
Additional information:

When I originally installed Gyroflow was getting an OpenCL error:

```
Failed to initialize OpenCL Platform::list: Error retrieving platform list: ApiWrapper(GetPlatformIdsPlatformListUnavailable(10))
```

I'm pretty sure Gyroflow was working just fine even with this OpenCL error - although there was an annoying startup delay. And to be clear: export was working just fine for me when I first ran Gyroflow.

Anyhow, I didn't like the startup error & delay and eventually discovered this OpenCL error can be solved by installing the `mesa-libOpenCL` package.

I don't know what I did to make the `NoCachedWgpuInstance` bug appear. Maybe I changed some setting, who knows. I tried removing both the `~/.cache/Gyroflow` & `~/.local/share/Gyroflow` folders to 'factory reset' the settings but that didn't fix the `NoCachedWgpuInstance` bug and broken export.

Going back to the start, just now I removed the `mesa-libOpenCL` package. Sure enough, the OpenCL error and startup delay returned, BUT... exports now work!

What is even more strange, re-installing the `mesa-libOpenCL` package now removed the OpenCL error and startup delay but exports still work.

I imagine on first run, there is some default setting that is applied when OpenCL is available that breaks export?

raintonr, 2024-12-06:
> I imagine on first run, there is some default setting that is applied when OpenCL is available that breaks export?

Yes! This is exactly the case.

In the settings that work, `"exportGpu-1": 0` is present. If I remove that, the `NoCachedWgpuInstance` reported occurs.

This settings item is not created by default when `mesa-libOpenCL` package is installed.

turtle-bazon, 2024-12-10:
Same for 1.6.0. My system is void linux. Disabling hardware encoding ("exportGpu-1": 0) doesn't solve my problem. 1.5.4 works fine.

turtle-bazon, 2024-12-10:
Found that chaning decoding devices solved problem. Actually not working OpenCL, all other types works.

[image]

AdrianEddy, 2025-01-18:
Fixed in cb594c9d1391f0c0d9b91ed3e4dfce7c4b6d9c9f

closed 2025-01-18 by AdrianEddy