gyroflow/gyroflow #935: Rendering time on the 1.6.0 is longer than 1.5.4

TierraErde, 2024-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.0

### What operating system are you using?

mac sonoma 14.6.1

### What GPU are you using?

M2 MAX

### What happened?

the same video, Rendering time on the 1.6.0 is longer than 1.5.4 about 100%,1.6.0 is 35fps/s.1.5.4 is 67 fps/s
this 1.5.4 log
[gyroflow.log](https://github.com/user-attachments/files/18052696/gyroflow.log)
i can't find the log of 1.6.0

AdrianEddy, 2024-12-08:
Try different "Device for video processing" and make sure you have both GPU encoding and GPU decoding selected

temp-64GTX, 2024-12-09:
i think it might be continuation that thing with ffmpeg. I also noticed that 1440p videos from gopro - is rendering much slower than 4k and 5k videos from gopro.

AdrianEddy, 2024-12-09:
Right, @TierraErde first of all try the dev build https://gyroflow.xyz/devbuild

TierraErde, 2024-12-13:
@AdrianEddy I'm sure both GPU encoding and GPU decoding selected

TierraErde, 2024-12-15:
yes,i try both dev build 1.6.0 and release 1.6.0

TierraErde, 2024-12-15:
and i try apple m2 max 1.2 encode with [openCL] apple m2 max decode;apple m2 max  encode with [openCL] apple m2 max decode;apple m2 max  1.2 encode with [wgpu] apple m2 max decode[metal],all of them the Rendering time is about 35fps/s~40fps/s

AdrianEddy, 2025-09-04:
I just compared 1.6.2 with 1.5.4 and 1.6.2 is significantly faster for me. 
Looks like they improved something in ffmpeg because there were no rendering changes in gyroflow itself