gyroflow/gyroflow #936: bug at 7:00

bygiants, 2024-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.0

### What operating system are you using?

Win 10

### What GPU are you using?

nVidia

### What happened?

Between 6:50 and 7:00 the image becomes pixelated :

Example 1
[image]
[image]

Example 2
[image]
[image]
from the video : 
https://www.youtube.com/watch?v=z358wOsBDAI&t=321s

lens profile : dji O3 
video speed : 33 %
h.265 - 200 MB/s

Hope that helps.
Please let me know if you need more info.

Thanks for sharing your work.
Have a great day.

Vincent Berger
bergervincent.fr

AdrianEddy, 2024-12-13:
try exporting with "GPU Decoding" (in Advanced)" disabled

AdrianEddy, 2024-12-21:
This should already be fixed in the dev build https://gyroflow.xyz/devbuild/ and if it's not fixed automatically, it's caused by the video decoder which is outside of my control. Exporting with GPU Decoding disabled should fix this

closed 2024-12-21 by AdrianEddy